realthunder/FreeCAD_assembly3 #804: Crash when closing transform

joanperelopez, 2021-09-11:
Hi
I am MAC user. I am using Mojave OSX.
The transform feature is working correctly, I can move, rotate, etc. but when I push the close button or push escape then Freecad app crash and close suddenly. Shown
[image]
 this dialog box:

luzpaz, 2021-09-11:
please always add full [About](https://wiki.freecadweb.org/About) info

realthunder, 2021-09-11:
Yeah, please post the About info. Also since its Mac, can you please dig up the crash report and post the crash stack? I've actually received a few similar reports from Mac user. But I can never reproduce it in my Mac, kind of frustrating.

thomasgi1, 2021-09-12:
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git)
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

I can reproduce the problem on my system (MacBook Air 13 inch 2017) with following steps:

- create a new document in PD
- create a sketch in XY plane
- draw a rectangle
- close the sketch
- pad the sketch
- apply the transform tool on the body (move the body in X and then rotate the body around X)
- close the transform tool
- crash:

Process:               FreeCAD [68385]
Path:                  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [68378]
User ID:               501

Date/Time:             2021-09-12 09:18:53.115 +0200
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Anonymous UUID:        00D0AAE4-4FA2-1AA2-1DC3-429829F4143B

Sleep/Wake UUID:       CDC10083-4121-45A4-847E-8E89F176BABD

Time Awake Since Boot: 480000 seconds
Time Since Wake:       2900 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [68385]

VM Regions Near 0x10:
--> 
    __TEXT                      109d2a000-109d32000    [   32K] r-x/r-x SM=COW  /Applications/FreeCAD asm3.app/Contents/Resources/bin/FreeCAD

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libCoin.80.0.1.dylib          	0x000000010d089e50 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.80.0.1.dylib          	0x000000010d089fbe SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.80.0.1.dylib          	0x000000010d103d12 SoDragger::~SoDragger() + 194
3   libFreeCADGui.dylib           	0x000000010a16d35e Gui::RDragger::~RDragger() + 14
4   libCoin.80.0.1.dylib          	0x000000010d26680c SoBase::destroy() + 172
5   libCoin.80.0.1.dylib          	0x000000010d259f1a SoBaseList::truncate(int) + 74
6   libCoin.80.0.1.dylib          	0x000000010d26ad01 SoChildList::~SoChildList() + 17
7   libCoin.80.0.1.dylib          	0x000000010d2c1ec5 SoGroup::~SoGroup() + 37
8   libCoin.80.0.1.dylib          	0x000000010d2dfb4e SoSeparator::~SoSeparator() + 14
9   libCoin.80.0.1.dylib          	0x000000010d26680c SoBase::destroy() + 172
10  libCoin.80.0.1.dylib          	0x000000010d259f1a SoBaseList::truncate(int) + 74
[... 764 lines cut ...]
Kernel Alloc Once                    8K        1 
MALLOC                           164.3M      167 
MALLOC guard page                   32K        8 
MALLOC_LARGE (reserved)            384K        1         reserved VM address space (unallocated)
OpenGL GLSL                        256K        3 
PROTECTED_MEMORY                     4K        1 
STACK GUARD                       56.0M        7 
Stack                             26.0M        7 
VM_ALLOCATE                      455.8M      315 
VM_ALLOCATE (reserved)           256.0M        2         reserved VM address space (unallocated)
__DATA                            57.7M      686 
__DATA_CONST                      20.4M      248 
__DATA_DIRTY                      1138K      155 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       551.4M      220 
__OBJC_RO                         70.2M        1 
__OBJC_RW                         2496K        2 
__TEXT                           654.2M      627 
__UNICODE                          588K        1 
mapped file                       94.6M       38 
shared memory                      752K       14 
===========                     =======  ======= 
TOTAL                              2.4G     2559 
TOTAL, minus reserved VM space     2.1G     2559

joanperelopez, 2021-09-12:
Hi realthunder.
I can't post the about information for now. I am not home where I have the desktop computer. I will post it tomorrow.
BR

joanperelopez, 2021-09-12:
Hi Thomasgi1. 
This is the exact case when crashes. But even when you place a primitive as a cube and you move by transform. When you close transform always crashes.
Thanks for reproduce.

thomasgi1, 2021-09-12:
I found an even simpler way to reproduce the crash. Just create an empty body within a new document and transform that body. The resulting stack trace is the same as I already posted here.

realthunder, 2021-09-12:
Yes, I've seen this crash in quite a few reports. Can you please try to turn on shadow draw style and tell me if you can see the shadow?

thomasgi1, 2021-09-12:
Yes, I see the shadow, if the body contains some kind of geometry. Of course, when the body is empty, then there is also no shadow.

realthunder, 2021-09-12:
Can you please send me your user conf file? You can export it using `Tools -> Edit parameters`. Make sure you are viewing the `<Application Settings>`, and select export. And send me the file.

[image]

thomasgi1, 2021-09-12:
Here it is...

thomasgi1, 2021-09-12:
Hmm, looks like replying to the github e-mail does not work for attachments. So, here is the conf file again...
[fc_application_settings.FCParam.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7149600/fc_application_settings.FCParam.zip)

joanperelopez, 2021-09-13:
Hi
This is my about post

[image]

Thomasgi1, I imported the applications settings you post, but the crash problems remain.

BR
Pere

domomib, 2021-09-13:
Same issue on Mac OS HighSierra

realthunder, 2021-09-15:
Should be fixed with the above commit. Will make new release soon. You can reopen if still got issue.

closed 2021-09-15 by realthunder