realthunder/FreeCAD_assembly3 #806: Can't move Task View/Property view outside main FreeCAD window

macdroid53, 2021-09-11:
I am using the 9/11 release, but this has been the case for as far back as I know.

I can't drag the dialogues like Task View and Property View to another monitor, outside the main window. This means I waste precious 3D view space on information dialogues and renders my multi-monitor un-helpful when using FreeCAD.

Note: in master this is not an issue. I can move these dialogues anywhere I desire.

OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

obelisk79, 2021-09-14:
@realthunder I too would like to see this behavior. There are many who use multi monitor setups. Being able to move the spreadsheet, and or panels outside of the main window to a second monitor seems like it would be pretty valuable for work-flow. I've been planning on adding a second monitor in the not too far off future as well.

closed 2021-09-15 by realthunder

macdroid53, 2021-09-21:
Sorry, this report is actually on the linkdaily, since I just got it to build. But, I figured two issues on the same thing would be confusing.

I can now move panels outside the main window.

But, once outside thay can only be resized with Alt-rightclick drag. It is apparently not responding to a mouse cursor on an edge for resize. The mouse cursor never changes to a drag arrow.

OS: Debian GNU/Linux 10 (buster) (XFCE/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28485 (Git)
Build type: Release
Branch: LinkDaily
Hash: 58e962438c54c632b5ab63481e832095a1701d63
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)