realthunder/FreeCAD_assembly3 #807: Some minor issues with 9/11 release

ngocminhdao88, 2021-09-11:
I am using the daily release on 11/09/2021 and notice some small issues with this version

The tabbar addon does not work any more. I need to re-enable the drop down menu to select workbench.
[image]

In clipping view with two clipping planes, some side of model does not show its edges. I am using draw style "As Is".
[image]

Is there a way for me to benchmark the performance of daily (with new renderer) vs stable version? I can feel that the daily version with new renderer is noticeably faster when loading big model, but by how much?

By the way, here is my FreeCAD info:

> OS: Ubuntu 20.04.2 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-09-12:
There is built-in support for showing workbench as tab bar. Just right click the drop down and select 'Expand'. You can drag and drop to reorder the tabs, and select which workbenches to appear with right click as well.

About the missing edge, is it possible for you to send me the file to have a look?

There is not much benchmarking done with the new renderer yet, as it is not matured yet. Right now the focus is more on the functional completeness comparing to the old rendering. You can use the frames per second as a rough benchmark.

[image]

ngocminhdao88, 2021-09-12:
Okay, so expanding the workbench drop down menu does work.

About the missing edges, here is just a simple cube for testing.
[simple_cube.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7149127/simple_cube.zip)

Due to the dark background is little hard to see the missing edges, but they are there.
[image]

Here are the settings for clipping tool. The cube is 50x50x50 mm. Some how the offset doesn't work correctly. Please double check.
[image]

I have two wishes for the clipping tool:

1. A ability to show and drag the clipping plane in 3D view.
2. Using a face of model as the clipping plane.

About the first wish. I some time visually lost the model in 3D view, if the offset did not set right. Without the visual cue, it is kind of hard to tell where i am in space.

Here is a a video about the clipping tool working in Fusion360
https://youtu.be/SFRMR_0CJXU?t=3434

As always, thank you for your works on FreeCAD project.

realthunder, 2021-09-13:
I see, you are using the 'Concave section' option. Unfortunately, this defect is an inherent limitation due to the rendering method used. I might find a way around in the future using other methods. BTW, the F360 video you posted is very nice. I'll definitely add that function later on.

herrgerd, 2021-09-17:
> 
> 
> I see, you are using the 'Concave section' option. Unfortunately, this defect is an inherent limitation due to the rendering method used. I might find a way around in the future using other methods. BTW, the F360 video you posted is very nice. I'll definitely add that function later on.

Using a face of the model (+/- offset) for clipping views would be awesome! Thanks for making that happen!!!

closed 2021-12-29 by realthunder