realthunder/FreeCAD_assembly3 #808: Stable & Daily crash at startup OS X Big Sur

tonealone, 2021-09-12:
Hello, 

I downloaded the latest Daily & Stable versions of FreeCAD but they both crash at startup.

This is the FreeCAD info from the other "HEAD" branch:

OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

here's the crash report when I start FreeCAD_asm3:
Process:               FreeCADLink [750]
Path:                  /Applications/FreeCAD.app/Contents/MacOS/FreeCADLink
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           FreeCADLink [750]
User ID:               501

Date/Time:             2021-09-11 09:11:04.635 -0400
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Anonymous UUID:        AF25D3EA-9AD4-61E5-D74D-B7BBB2198A67

Time Awake Since Boot: 40 seconds

System Integrity Protection: disabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  dyld: Using shared cache: 0C9EB0DE-F5CE-3AFC-8F08-0C6DB1E7FC44
Library not loaded: /usr/local/opt/open-mpi/lib/libmpi.40.dylib
  Referenced from: /Applications/FreeCAD.app/Contents/MacOS/FreeCADLink
  Reason: image not found

Binary Images:
       0x10c8c0000 -        0x10c8cffff + (??? - 2021) <5F1001CC-CFAC-3F2B-B981-8A768BAB7A3A> /Applications/FreeCAD.app/Contents/MacOS/FreeCADLink
       0x10cf43000 -        0x10cfdefff  dyld (852.2) <57DB2053-BFD5-3683-97C6-F1DB2A1F1D09> /usr/lib/dyld

zardozerr, 2021-09-12:
I can confirm that I get the same thing, Big Sur 11.5.2, on a 2019 iMac (Intel). Anyone know of a way to get the 7.17 builds in the meantime? Seems they've been taken off the releases, so I can't download them.

Here's the crash log:
Process:               FreeCADLink [2347]
Path:                  /Applications/FreeCAD Linkstage3/FreeCAD.app/Contents/MacOS/FreeCADLink
Identifier:            ???
Version:               ??? (2021)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           FreeCADLink [2347]
User ID:               502

Date/Time:             2021-09-12 10:16:01.370 -0700
OS Version:            macOS 11.5.2 (20G95)
Report Version:        12
Anonymous UUID:        023EFDD0-5BA1-084C-E097-7995304D4DB0

Sleep/Wake UUID:       4BF630F5-B211-43A7-924E-F292FAABE798

Time Awake Since Boot: 3700 seconds
Time Since Wake:       1800 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  dyld: Using shared cache: 0C9EB0DE-F5CE-3AFC-8F08-0C6DB1E7FC44
Library not loaded: /usr/local/opt/open-mpi/lib/libmpi.40.dylib
  Referenced from: /Applications/FreeCAD Linkstage3/FreeCAD.app/Contents/MacOS/FreeCADLink
  Reason: image not found

Binary Images:
       0x1026a8000 -        0x1026b7fff + (??? - 2021) <DAEEA1E4-D913-31F4-B81D-1611BB7976CD> /Applications/FreeCAD Linkstage3/FreeCAD.app/Contents/MacOS/FreeCADLink
       0x108ed4000 -        0x108f6ffff  dyld (852.2) <57DB2053-BFD5-3683-97C6-F1DB2A1F1D09> /usr/lib/dyld

Model: iMac19,1, BootROM 1554.140.20.0.0, 6 processors, 6-Core Intel Core i5, 3 GHz, 8 GB, SMC 2.46f13
Graphics: kHW_AMDRadeonPro570XItem, Radeon Pro 570X, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/ChannelA-DIMM0, 4 GB, DDR4, 2667 MHz, SK Hynix, HMA851S6CJR6N-VK
Memory Module: BANK 2/ChannelB-DIMM0, 4 GB, DDR4, 2667 MHz, SK Hynix, HMA851S6CJR6N-VK
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x7BF), wl0: Jul  7 2021 02:34:10 version 9.30.444.17.32.5.70 FWID 01-4075fabc
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE HDD ST1000DM003, 1 TB
USB Device: USB 3.1 Bus
USB Device: BUP Ultra Touch
USB Device: Magic Keyboard
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB Optical Mouse
Thunderbolt Bus: iMac, Apple Inc., 47.5

realthunder, 2021-09-13:
@zardozerr I just uploaded some older version [here](https://drive.google.com/drive/folders/19eIWuQGtKR3GMQA66xBstoQ43b_5LFGT?usp=sharing). Are you guys using the conda build?

zardozerr, 2021-09-13:
@realthunder thanks for the quick reply. No, I'm trying both the Stable and Daily Big Sur builds for this particular machine, and both those builds come up with similar crashes. I'll be able to test the conda build on a machine with Catalina in my office tomorrow, and I can report of there's a problem with that also.

zardozerr, 2021-09-14:
The regular Conda builds, both Daily and Stable, work normally on Catalina, so it seems this is limited to the Big Sur builds.

NothanUmber, 2021-09-15:
The non-Big Sur build seems to work on Big Sur (x86) on first sight. Apparently the Big Sur build has some dependencies to OpenMPI which isn't installed by default. So it might be possible to get this to run by installing the needed version of OpenMPI, e.g. via ports. But if this dependency isn't needed it might be better to rebuild the binary without it (or bundle the required lib).

Would be good if old releases wouldn't automatically be deleted when a new version comes out. (E.g. 20210717 would have been a good fallback to keep)