realthunder/FreeCAD_assembly3 #809: 9/11 release issues: dialog box crash, numpy

alowell-ssl, 2021-09-12:
Ubuntu 20.04 LTS, both stable and daily have the same issue.

Steps to reproduce:

1. Start Freecad
2. Click "create new"
3. Switch to part design WB 
4. Click "create body"
5. Click "create sketch"
6. Double click sketch in tree view
7. Dialog box comes up that says "a dialog is already open in the task panel, do you want to close this dialog?".  Click yes.
8. Freecad segfaults, I get this on the terminal:

Wizard shaft module cannot be loaded
Program received signal SIGSEGV, Segmentation fault.
/tmp/.mount_FreeCAzusWhD/AppRun: line 46: 32587 Segmentation fault      (core dumped) "$PREFIX/bin/$name" "$@"

--------------

ALSO I get a lot of error messages when starting freecad, regarding numpy.  Here is what I get on the terminal:
Path WB not found                                                                                                                                                                                                  
During initialization the error "                                                                                                                                                                                  
                                                                                                                                                                                                                   
IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!                                                                                                                                                 
                                                                                                                                                                                                                   
Importing the numpy C-extensions failed. This error can happen for                                                                                                                                                 
many reasons, often due to issues with your setup or how NumPy was                                                                                                                                                 
installed.                                                                                                                                                                                                         
                                                                                                                                                                                                                   
We have compiled some common reasons and troubleshooting tips at:                                                                                                                                                  
                                                                                                                                                                                                                   
    https://numpy.org/devdocs/user/troubleshooting-importerror.html                                                                                                                                                

Please note and check the following:

  * The Python version is: Python3.9 from "/tmp/.mount_FreeCAzusWhD/usr/bin/FreeCADLink"
  * The NumPy version is: "1.20.2"
and make sure that they are the versions you expect. 
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy.core._multiarray_umath'
" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/alowell/python_environments/py3/lib/python3.8/site-packages/numpy/core/__init__.py", line 22, in <module>
    from . import multiarray
  File "/home/alowell/python_environments/py3/lib/python3.8/site-packages/numpy/core/multiarray.py", line 12, in <module>
    from . import overrides
  File "/home/alowell/python_environments/py3/lib/python3.8/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ModuleNotFoundError: No module named 'numpy.core._multiarray_umath'

During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "<string>", line 158, in InitApplications
  File "/tmp/.mount_FreeCAzusWhD/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/tmp/.mount_FreeCAzusWhD/usr/Mod/asm3/freecad/asm3/init_gui.py", line 2, in <module>
    from . import gui
  File "/tmp/.mount_FreeCAzusWhD/usr/Mod/asm3/freecad/asm3/gui.py", line 10, in <module>
    from .utils import getElementPos,objName,addIconToFCAD,guilogger as logger
  File "/tmp/.mount_FreeCAzusWhD/usr/Mod/asm3/freecad/asm3/utils.py", line 11, in <module>
    import numpy as np
  File "/home/alowell/python_environments/py3/lib/python3.8/site-packages/numpy/__init__.py", line 145, in <module>
    from . import core
  File "/home/alowell/python_environments/py3/lib/python3.8/site-packages/numpy/core/__init__.py", line 48, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/tmp/.mount_FreeCAzusWhD/usr/bin/FreeCADLink"
  * The NumPy version is: "1.20.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy.core._multiarray_umath'

realthunder, 2021-09-12:
The crash has been fixed with the above commit.

The Numpy problem is caused by conflicting Python version of your local site-package (3.8) and the one running in the AppImage (3.9). I am not sure how to solve this. I'll need to search around for some answer. 

PS. next time, please try to split the issue for each problem you have. Thanks!

closed 2021-09-12 by realthunder

realthunder, 2021-09-13:
Hi @looooo, could you please take a look at the above Numpy version mismatch problem? Any idea on how to solve it?