realthunder/FreeCAD_assembly3 #811: moving tool should be created like linearPattern tool . Angle transformations tool should also be added like polarPattern tool..

LeoBinsc, 2021-09-12:
moving tool should be created like linearPattern tool . 
Angle transformations tool should also be added like polarPattern tool.

moving tool  have function of choosing axis at any angle , which can be selected or filled in.   like (1,0,0) or others.
The icon of moving tool should be a coordinate system.

Angle transformations tool have function of filling angle in Write box,  and select a Datum plane at first.
The icon of angle transformations tool should be a Oblique coordinate system.
[image]
[image]
[image]
[image]

mwlandolsi, 2021-09-13:
I think that these tools are available in draft WB, no ?

LeoBinsc, 2021-09-14:
my first picture shows it is Auxiliary，because that pattern is not easy to create by revolve tool.
so i think it is a  Auxiliary tool. some time it is easy to create Complex pattern.

LeoBinsc, 2021-09-14:
i have not used draft WB before.

LeoBinsc, 2021-09-14:
Select the object and fill in the offset of X, y and Z

Or move 3 times along the X, y and Z axes respectively, and refresh each time.

Point coincidence  movement,  Line coincidence  movement and coplanar(in same plane) movement are equally important

LeoBinsc, 2021-09-14:
Select one object and fill in the offset of X, Y and Z.

Or move 3 times along the X, y and Z axis respectively, and refresh when change then.

Point coincidence  movement,  Line coincidence  movement and coplanar(in same plane) movement are equally important

LeoBinsc, 2021-09-14:
i put a work, which shows the two part of one body have no point to contact each other.
[part design026.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7162224/part.design026.zip)