realthunder/FreeCAD_assembly3 #813: incorrect geometry with variant links of PartDesign bodies

alexneufeld, 2021-09-12:
This ONLY occurs with the 2021/09/11 Daily build, as far as I can tell.

[image]

To reproduce:
- create a partdesign body that uses multiple separate solids to build up a part, then boolean cut one of the bodies from the other
- specify a Property in the body (IE Body.MyDimension = 20mm), then use href to refer to that property in one of the bodies geometric elements. Set the CopyOnChange bit for this property
- Create a link to the body, set CopyOnChange to Enabled, change the custom property
- The variant link will have incorrect geometry 

[image]

[image]

OS: Manjaro Linux (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Canada (en_CA)

[variant_link_geom_error.ZIP](https://github.com/realthunder/FreeCAD_assembly3/files/7150058/variant_link_geom_error.ZIP)
(change file extension to .FCStd)

oficinerobotica, 2021-09-14:
I can confirm the strange behavior of variantLink  objects in the latest daily. In the screenshot below, the protruding rods are actually intended to be pockets. 
[image]

OS: KDE neon User Edition 5.22 (KDE/plasma)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4128 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 0337073f17a1128e06e32a4bdce2e51f80b9f23a
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

realthunder, 2021-09-14:
Wow, this is disastrous bug! I have pull the current releases. Will make a new one very soon. Thanks for reporting.

BTW, it is not the variant link that causes the problem. It is the Pocket feature initialization got problem, causing its AddSubType secretly becoming 'Additive'. I think there are other features affected by this problem. Do not use the current release. Wait for my updates.

closed 2021-09-14 by realthunder