realthunder/FreeCAD_assembly3 #814: How to Highlighting sheet metal bends in construction tree

mwlandolsi, 2021-09-13:
Hello,
I do not know if this problem remains to FreeCAD or to sheet metal WB, but when moving mouse over the bend, or on click, the correspondent  name in the tree does not highlight.
this is very useful when modifying a bend length of a shape that has many bends. in that case it will be difficult to select which bend to select in the tree.
[image]
[image]
[image]

luzpaz, 2021-12-27:
Please ask on the [official sheetmetal forum thread](https://forum.freecadweb.org/viewtopic.php?f=3&t=60818)

mwlandolsi, 2021-12-30:
I will transfer it to the forum, Thank you

luzpaz, 2022-03-18:
@mwlandolsi please follow-up did you post to the forum, if so can you x-post the URL? And also share what the outcome was, please? thanks!

mwlandolsi, 2022-03-19:
Thanks for your answer. Yes,  a post has been submitted to the forum and this is the link : https://forum.freecadweb.org/viewtopic.php?f=20&t=65062&p=559234#p559234

the last post was  : 
by [Pauvres_honteux](https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=2803) » Thu Jan 06, 2022 9:40 am

    [mwlandolsi](https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=39759) wrote: [↑](https://forum.freecadweb.org/viewtopic.php?p=559234#p559234)Wed Jan 05, 2022 7:17 pm
    ... sometimes we need to click or hi-light a function (bend) on the view or on the tree ...

Hi mwlandolsi, this sounds like it could be in the same ballpark as: [Ticket #4730 Children and/or parent to be highlighted when hoovering over something in the tree/3D viewport](https://forum.freecadweb.org/viewtopic.php?f=34&t=57545&sid=35cc4126ec77ced93f349bb53ca03ccb)