realthunder/FreeCAD_assembly3 #816: menus in white

joanperelopez, 2021-09-13:
[image]

Go to View -> toolbars or View -> Panels and you will see:

[image]

[image]

BR
Pere

realthunder, 2021-09-14:
Which stylesheet do you use? If it's customized, please post it here. Thanks

thomasgi1, 2021-09-14:
I think, the problem is that a style sheet is used at all for the Panels and Toolbars sub-menus. If look at the video below, then you see a clear difference in rendering the Workbench sub-menu and the Panels or Toolbars sub-menus.
My style sheet settings are:
- Style sheet: No Style Sheet
- Overlay style sheet: Light-Outline
- View menu style sheet: Light

https://user-images.githubusercontent.com/17226631/133259378-f5df7ec2-0552-4806-b225-62984229beda.mov

closed 2021-09-15 by realthunder

joanperelopez, 2021-09-15:
Solved!
I changed to "No Style Sheet" and it works correctly
Thanks a lot

realthunder, 2021-09-15:
Do you compiled my branch yourself? Anyway, it is not really solved. I find that MacOSX has trouble showing a checkbox widget in its global menubar. So I just disable it on MacOSX specifically. The one showing now is a normal checkable menu item. The difference is that for normal checkable item, the menu will immediately close when you click any item, while for checkbox, you can click multiple items, which is really convenient when you want to toggle multiple toolbars for instance. You still get the access to the checkbox menu if you right click any toolbar.