realthunder/FreeCAD_assembly3 #817: [Bug] Can't read properties of features

YosypC, 2021-09-14:
In relation to #815 

[image]
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git)
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)
```

[image]
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3842 (Git)
Build type: Release
Branch: LinkDaily
Hash: 44be6d539871be4083e83f428e8fcfbcf38408ef
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)
```

realthunder, 2021-09-14:
The auto complete feature has been changed a bit. You need to type some character to trigger the search. The auto complete now returns match keyword contained in any position, instead of just match from the beginning.

YosypC, 2021-09-14:
Would it be possible to order the results based of types, just like it was before?

closed 2021-09-30 by realthunder