realthunder/FreeCAD_assembly3 #818: [Feature request] Automatically apply symmetry constraints to geometry created with symmetry tool

YosypC, 2021-09-14:
Title is self explanatory. Newly created geometry with the symmetry tool appear green, while, in fact, does not apply any symmetry constraint.

```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git)
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)
```

wsteffe, 2021-09-18:
Hello YosypC, 
 I agree with you that proper symmetry constraints should be automatically added to a shape created with the symmetry command of the Sketch WB. 
But I think that this is probably an upstream issue and that this post should be addressed to abdullah (in the FC forum) which is the main developer of the Sketch WB.

YosypC, 2021-09-18:
You're right, I made a [post](https://forum.freecadweb.org/viewtopic.php?f=8&t=62308) on the forum. Feel free to join.

closed 2021-09-18 by YosypC