realthunder/FreeCAD_assembly3 #820: [Feature] Symmetry Constraint workflow improvement

BloodyRain2k, 2021-09-15:
I'm not sure how others use the Symmetry Constraint, but for me it's at least half of the time to be able to use diameter measurements in circular profiles.
Reason for that is that my brain can easier connect to the value if it's what I'm using, instead of seeing it converted into a radius.

Meaning, most of the time I just want to mirror a point across the vertical axis and then distance constraint them.

So I was wondering, if it would be up to consideration if the Symmetry Constraint could create the 2nd vertex itself, if there's only 1 edge and 1 vertex selected.
Currently that just throws an error about fixed geometries.

Following that, could it maybe also keep both vertices selected after creating the constraint, so it would be possible to directly apply a distance constraint without needing to re-select them first?

PS: I'm also thankful to anyone, who could point me at the neccessary sources to learn about the functions and operations needed to create a macro for this. Because either way, I'd need that at the least temporarily : /

herrgerd, 2021-09-17:
I guess you mean this, right? 
https://www.cati.com/blog/2019/12/solidworks-add-diametric-dimensions-to-revolve-sketches/

I improvise on this lots of times by just creating a mirrored reference line. My brain works exactly like yours, it doesn't want to see a radius dimension where I truly want to see a diameter...

BloodyRain2k, 2021-09-17:
Well, that would be the "endgame" solution.
For now I'd be happy if I could be spared having to first create the 2nd point and then not having to reselect both points for the distance constraint.
That would already improve the workflow for this usecase a lot!

I tried myself at creating a macro for that, but I'm stuck at converting the Gui.Selection information from geometry names to sketcher ids needed for the constraints : (