realthunder/FreeCAD_assembly3 #821: Shared toolbars (draft snap, Measure) misbehaving

StudioPetrikas, 2021-09-15:
Hi,
Incredible work on the UI in the latest release. Thank you so much for that.

However, while making custom configurations, I have noticed that the toolbars that are shared throughout workbenches (toolbars like "draft snap" or "Measure") won't keep their place and they ignore being enabled/disabled. 

I can replicate this on a clean/fresh setup.

Here's a short video showing the behaviour:
https://www.youtube.com/watch?v=320eW06CrAw

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.911.24301 +4130 (Git)
Build type: Release
Branch: LinkDaily
Hash: cdcdcfc78188218bd5eab1cfd9f911acd666f734
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)

Thanks!!

realthunder, 2021-09-22:
This is a tough problem to solve. I blame this for Qt's QMainWindow API, which makes impossible to get complete control of its toolbar layout.

What is currently done is that when switching workbenches, toolbars are not added or removed, but instead only gets shown or hide. QMainWindow maintains two different layouts. The internal layout that holds all docked toolbars, and another layout for the visible toolbars, which is what the end-user sees. So the first visible column of the toolbars may actually be the second or third internal columns. And when you move a toolbar to an existing column, it may in fact be appended to many hidden toolbars from other workbench. That's why it may seems that the shared toolbar jumped to a new position when switching workbench. It's actually just the hidden toolbars become visible. However, QMainWindow has no API to control exactly where the toolbar to be inserted.

Before I can finally come up with a solution (which I doubt it myself), there is some kind of work around. That is, to move the shared toolbar to a different docking side of the workbench exclusive toolbars. This is the easiest way. Or, if you insist on docking the toolbar at the same side, you need to make sure to move it to a new row/column. So, if you are docking to the right side of the main window, then the row is counting from right to left, and moving the toolbar to the left gets you a new row, while moving it to any existing column may result in jump as there are likely other hidden toolbars reside in that column. You can then manually shuffle the toolbar to the desired column, but need to do it in each workbench it appears.

[image]

StudioPetrikas, 2021-09-22:
Okay, we can work around that. Perfectly understandable that you are limited by Qt.
Unrelated, but just a side note, I don't remember having this issue with Persistent toolbars plugin (which wrote the toolbar arrangement in plain text in the cfg, genius!), maybe there's something that can be used from it?

This might be a great time to potentially address a different UI problem - the status bar. It takes a full line along the screen and uses a fraction of that space for good information. The shared Measure and Snap Tools would fit perfectly there, as right now the status bar is pretty pointless (New tooltips showing all the necessary information, "Navigation Bar" is something you set-and-forget)

Is it possible to make a toolbar area on the Status Bar without spending a tone of effort?

Thanks!