realthunder/FreeCAD_assembly3 #823: Creating pocket with fit tolerance requires base sketch to be recomputed

bennvan, 2021-09-16:
When creating a simple pocket with the 'fit tolerance' setting, sometimes the base sketch needs to be recomputed to generate the geometry correctly, and I'm not sure why this is the case. 

I am afraid if geometry fails on a more complicated model it will be hard to know if the fix might be something as simple as regenerating a sketch somewhere.

[image]

OS: Ubuntu 20.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.916.24301 +4152 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 7fd36f96f4db17db4f8d058eab22588cca8f9d5c
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Australia (en_AU)

realthunder, 2021-09-29:
Can you please provide a sample file showing the problem?

bennvan, 2021-09-29:
Attached is the file just after the pocket feature is applied but before recomputing. I have also noticed the problem does not occur when applying a pocket in the same way to a cylinder.

[ModelBeforeRecompute.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7248502/ModelBeforeRecompute.FCStd.zip)
d

closed 2022-01-31 by bennvan