realthunder/FreeCAD_assembly3 #824: [BUG 2021.9.16] Sketcher: picking priority issues when doing constraints

Miksu80, 2021-09-16:
When doing for example point-symmetry, the symmetry icon prevents clicking on vertex. It'd be good to disable constraint selection when constraint is being added. Basically, if you start to do constraints by clicking the constraint-button 1st, then it wouldnt try to select constraints at all but just vertices, lines, arcs, construction geometry, axis etc.) This is a new issue that has not existed on few months back version.

Here trying to select the point with coincident -tool and it just tries to select the constraint:
[image]

It does this some times on center point but if the sketch plane is on XY,YZ, XZ plane, the center-point is ontop but if you do a sketch in a surface which is ontop of sketch center, center-point goes behind constraint. Even with "GG" it doesn't work:
[image]
with just selecting the vertex 1st or 2nd, it only highlights it, it doesnt really select it:
[image]
When using ctrl, it highlights both but just dont work:
[image]

Version:
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.916.24301 +4152 (Git)
Build type: Release
Branch: LinkDaily
Hash: 7fd36f96f4db17db4f8d058eab22588cca8f9d5c
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

herrgerd, 2021-09-16:
I'm 100% with you here. The `GG` command (which could be super super useful) is - in its recent form - totally useless. The other "bug", namely selecting constraints, has been mentioned many many times before, too.

I hope both will get some attention soon. Sketching, _which is the start of every part or assembly_, can still be a pain because of that...

closed 2021-09-17 by realthunder

realthunder, 2021-09-17:
I have fixed the symmetric constraint z order problem. But I don't quite get it what you mean by it just doesn't work when using ctrl + `G, G`?  Can you please elaborate?

Miksu80, 2021-09-17:
When trying to select the "hidden" vertex with G, G it doesn't really select it. With constraint tool clicked, it didn't select the vertex using G, G and selecting the vertex from the list.

Miksu80, 2021-09-29:
The dimension texts are in front of geometry when doing symmetry (or other constraints) -this is not a biggie but noticed with small diameter circles (had to zoom in to be able to select center point, didn't test "G, G").
[image]

realthunder, 2021-09-29:
The constraints having datum label are giving higher priority because you can move them around, while other constraints cannot. So, why not just drag it away?

Miksu80, 2021-09-29:
> The constraints having datum label are giving higher priority because you can move them around, while other constraints cannot. So, why not just drag it away?

That works also, just didn't notice to move the number prior to selecting tool. When tool already selected, you probably don't select anything other than geometry so basically anything but geometry can be filtered out when tool is already selected. Anyways it's not a biggie and can be dealt with your suggestion or by zooming in 👍