realthunder/FreeCAD_assembly3 #826: Segmentation fault on start (but works if set `$ LIBGL_ALWAYS_SOFTWARE=1`)

Symbian9, 2021-09-16:
- **Version:** 2021.09.16 Daily (AppImage)
- **OS:** Linux / antiX-21-beta2 (based on Debian 11)

### Expected behaviour

App should run without even `$ LIBGL_ALWAYS_SOFTWARE=1`.

### Actual behaviour

It fails on start:

```
$ ~/Downloads/FreeCADFreeCAD-asm3-Daily-Conda-Py3-Qt5-20210916-glibc2.AppImage
/tmp/.mount_FreeCANND5i0
FreeCAD 0.19, Libs: 2021.916R24301 +4152 (Git)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bd60) [0x7fa73afabd60]
```

But if launch it with `$ LIBGL_ALWAYS_SOFTWARE=1` then it starts and works (but slowly).

realthunder, 2021-09-17:
Can you please paste the full crash call stack?

Symbian9, 2021-09-17:
> Can you please paste the full crash call stack?

What should I do to get full stack?

JFTR, Same issue also happen with *official* FreeCAD AppImage.

realthunder, 2021-09-17:
> #0  /lib/x86_64-linux-gnu/libc.so.6(+0x3bd60) [0x7fa73afabd60]

Are there anything printed out following this line? Anything special about your display driver? Sounds like a driver issue.

Symbian9, 2021-09-17:
> Are there anything printed out following this line?

Nope. Thats all.

FTR, It is antiX Linux distro:

- https://sourceforge.net/projects/antix-linux/files/Testing/
  - https://antixlinux.com/antix-21-beta2-iso-files-available-debian-11-based/