realthunder/FreeCAD_assembly3 #827: [Enhansment] fillet between arc and line

mwlandolsi, 2021-09-16:
Hello,
It is very common to create a fillet between an arc and a line in mechanical design. This could be made in sketch or after generation the 3D.
I recently discovered that creating a fillet between an arc and a line is not possible in all Freecad versions 0.19. it is possible to create an arc and make it tangent to the arc and the line. But why this is not possible by using the fillet button in sketch ?
[image]

herrgerd, 2021-09-17:
[image]

Like so?

mwlandolsi, 2021-09-17:
Oh yes, thank you, I confused with fillet between tow lines, when we click on the intersection-end point, it is possible to get the fillet. but between an arc and a line it looks like you have mentioned above. It could be a UX enhancement to be able to select the point of intersection of the line and the arc  (may be other types of lines) where to place the fillet.
https://user-images.githubusercontent.com/76944804/133747925-267df6e6-27b0-43a6-9f7e-5811ff0c596c.mp4

herrgerd, 2021-09-17:
Ah, now I see what you mean. Right, that would be nice if it worked, too.
The method used above has one advantage, though: You can immediately decide how big the fillet will be by clicking closer/farer away from the intersecting point. That's neat!