realthunder/FreeCAD_assembly3 #831: FreeCAD on Linux Mint: Instant crash when hitting new sketch button.

Geitde, 2021-09-16:
FreeCAD is now crashing on sketch creation like it does in the official 0.19 branch.

Using latest Linux Mint. Tested with 16th september

Clicking on new sketch crashes FreeCAD instantly.

[FreeCAD.crashlog.txt](https://github.com/realthunder/FreeCAD_assembly3/files/7181472/FreeCAD.crashlog.txt)

realthunder, 2021-09-21:
Looks like a driver issue. Is this repeatable? Did it occur on earlier release?

Geitde, 2021-09-22:
It seems to be triggered by a Mint update happened around two month ago, as all old versions are not effected, BUT I don't think it is a driver issue.

If I load an old FreeCAD model I can open sketches and modify them. Just creating a new one is causing a crash.

luzpaz, 2022-01-19:
@Geitde can you link to downstream Linux Mint package discussion? If you need a workaround in the meantime, you can use the [AppImage](https://wiki.freecad.org/AppImage) or [Snap]((https://wiki.freecad.org/Snap)

luzpaz, 2022-03-19:
@Geitde are you aiming to use @realthunder's Link branch or are you using FreeCAD master branch? 

If the former, there is a Snap package for realthunder's Link branch: https://snapcraft.io/freecad-realthunder
If the latter there is a [Snap](https://wiki.freecad.org/Snap) package for FreeCAD master

Geitde, 2022-03-23:
I only use Linux Mint and the provided appimages. The official ones and the real thunder ones. I prefer the RealThunder one, as it is the way to go and it should have been merged into the official branch a long time ago without all the fuzz they made. It just works perfect. 

On my main desktop system both versions are crashing, while on my laptops with the same operation system version they work. So it really seems to be a driver issue triggered by FreeCAD on this one system and only exact the moment when the sketch gets created.

I can select new sketch and the plane I want it on and when I confirm e.g. x/y plane it prepares to render the empty sketch and an exception is terminating the application.

Fun part is that I can use and open any sketch created on other systems by simply loading the FCStd file. 

I will probably stuff some gfx card into my systems currently only the internal intel gpu is used, which works fine for all my everyday usage. Currently it is just inconvenient to always swap to a dedicated FreeCAD device instead of using the system I always used.

luzpaz, 2022-03-23:
@Geitde this is beside the point, I'm trying to explain that there are [realthunder provided snap packages](https://snapcraft.io/freecad-realthunder) as well.

luzpaz, 2022-03-23:
Can you share more about your specs and driver ?

Geitde, 2022-03-23:
[image]