realthunder/FreeCAD_assembly3 #833: my macro not auto starting? since changed to 2021/09/16

bambukouk, 2021-09-17:
I am using (all the time) a macro created by mario52.
It was working fine until I updated to FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210916

It will not autostart when I open FreeCAD.
I can start it manually, but it is a bit of a PITA.

I am sure it is probably trivial for someone who knows what they are doing (but that's not me...).
It's all there and unchanged and parameter editor is still OK:

[image]

so perhaps someone can help me, please?

```
here is the only bit of the macro that is likely to cause problem:
import FreeCAD, FreeCADGui
App = FreeCAD
Gui = FreeCADGui

import PySide2
from PySide2 import (QtWidgets, QtCore, QtGui)
from PySide2.QtWidgets import (QWidget, QVBoxLayout, QStyle, QDockWidget, QToolBar, QDialog)
from PySide2.QtGui import (QColor, QIcon)
from PySide2.QtCore import QSize

global switchConventionalView # switchConventionalView 0, 1
global switchToolbarIconSize    # switchToolbarIconSize 0, 1 
                                # 0 = size icon FreeCAD parameters
                                # 1 = size on choice
global sizeIconX                # size X icon (if switchToolbarIconSize = 1)
global sizeIconY                # size Y icon (if switchToolbarIconSize = 1)
global setFCIconModel           # 0 = FreeCAD icon fromTheme
                                # 1 = Locomotive icon
                                # 2, 3, .... other only Text
global setStyleSheetToolBar     # 0 = not setStyleSheet
                                # color in HTML format ex: "#F8E6E0" (by default)
global switchNotRunAuto         # 0 = run automatic            ## is include in initGui.py file (change only in user parameter)
                                # 1 = not run the macro        ## is include in initGui.py file (change only in user parameter)
```

realthunder, 2021-09-17:
Can you please provide a link to the macro so that I can try?

bambukouk, 2021-09-18:
Thank you realthunder! 👍 
Macro below:

[ToolViewReverse_ToolBar_22-10-2020_b.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7189455/ToolViewReverse_ToolBar_22-10-2020_b.zip)

realthunder, 2021-09-18:
The required modification of ToolViewReverse_ToolBar.FCMacro is as follow. Since ToolViewReverse_ToolBar.py is exactly the same as .FCMacro file, it needs the same modification.

Basically, my branch changes how toolbar position are preserved by always pre-creating all toolbars that has ever appeared in FreeCAD, so that Qt can save and restore the exact position of these toolbars. The pre-created toolbars has the same object name as before, but no content. So for existing toolbar related addon or macro, instead of checking if the toolbar with a given name exists, it shall check if the toolbar has any content, or as a shortcut, if the toolbar has the title.

```diff
@@ -1120,24 +1120,27 @@
 mw = FreeCADGui.getMainWindow()
 tb = mw.findChildren(QtWidgets.QToolBar)
 
-switchtbar = 0
+switchtbar = None
 
 for i in tb:
     if str(i.objectName()) == "MyViewTBar":
-        FreeCAD.Console.PrintError("MyViewTBar" + " is already open" + "\n")
-        switchtbar = 1
+        switchtbar = i
 #        if i.isVisible():
 #            i.setVisible(False)                    # visible False
 #        else:
         i.setVisible(True)                     # visible True
         break
 
-if switchtbar == 0:
+if switchtbar == None:
     toolbar1 = QtWidgets.QToolBar()
     toolbar1.setObjectName("MyViewTBar")
+    mw.addToolBar(toolbar1) 
+else:
+    toolbar1 = switchtbar
+
+if not toolbar1.windowTitle():
     toolbar1.setWindowTitle("MyViewTBar")
     
-    mw.addToolBar(toolbar1) 
     toolbar1.setIconSize(QSize(sizeIconX, sizeIconY))
 
     try:
```

[ToolViewReverse_ToolBar.FCMacro.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7189625/ToolViewReverse_ToolBar.FCMacro.zip)

bambukouk, 2021-09-18:
Thank you 👍 
however, (unless I got things wrong...) your attached macro is no different from original?
or was I supposed to modify the macro using the code you have suggested?
I still have the same problem - sorry for being so thick.

realthunder, 2021-09-18:
The attached script is the modified one.

bambukouk, 2021-09-18:
I have copied ToolViewReverse_toolbar.FCMacro into C:\Users\bambuko\AppData\Roaming\FreeCAD\Macro directory.

and the same file (renamed ToolViewReverse_toolbar.py) into C:\Users\bambuko\AppData\Roaming\FreeCAD\Mod\ToolViewReverse_ToolBar directory.

Then tried openning FreeCAD - no sign of "my" toolbar :-(
tried openning macro manually - toolbar works fine.
Tried openning FreeCAD again - and again no sign of "my" toolbar :-(

Am I doing something wrong?

realthunder, 2021-09-19:
Yeah, there is indeed some incompatibility introduced in the current release. I'll fix that in the next one. Meanwhile, try the script below for a workaround. It uses a timer to force showing the toolbar. This is not the recommended way, but should be able to do its work for the time being.

[ToolViewReverse_ToolBar.py.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7190864/ToolViewReverse_ToolBar.py.zip)

bambukouk, 2021-09-19:
Thank you very much, indeed 👍

closed 2021-09-21 by realthunder

bambukouk, 2021-09-24:
@realthunder 
just a sanity check, please
#47522a6 is not yet part of the latest release (2021.09.22)?
 
because I have tried the macro you have modified for me (https://github.com/realthunder/FreeCAD_assembly3/issues/833#issuecomment-922245521) and it is still not working as it should ...

In the meantime I am using "...not the recommended way..." script (https://github.com/realthunder/FreeCAD_assembly3/issues/833#issuecomment-922399632)

sorry to be such pain