realthunder/FreeCAD_assembly3 #834: [Issue] Strange edge not supposed to be present, rendering chamfer and fillets impossible (BRep_API: command not done)

YosypC, 2021-09-17:
[image]

Normally no edge would be present if things are done well, in the blue circle.
Chamfering or filleting the Black edge is no problem.

[image]

However, in my model, all of this is not possible.
Highlighted in blue is a strange edge, which does not allow me to chamfer the red one, giving me the error in the title unless I select the adjacent edges (Black one)
I am sure I've done the pad right; I've tried padding perpendicularly from the face, and from the ground up. Still, nothing.

Model: https://1drv.ms/u/s!AqOOVg5zwVGuh6Rj--CoCSOKPM058g?e=0Sb1p7 : pasting OneDrive because GitHub is giving me a "is not included in the list" error

realthunder, 2021-09-18:
It's most likely an OCC issue. Here is a simplified shape showing the problem. The same file works in OCC 7.4, but failed in 7.5.X.

[image]

[draftfail.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7189462/draftfail.FCStd.zip)

YosypC, 2021-09-18:
How can this problem be solved?

realthunder, 2021-09-19:
I have isolated the problem inside OCC. It is due to a patch submitted by one of the forum member. See discussion [here](https://forum.freecadweb.org/viewtopic.php?f=3&t=54353&p=534260#p534260). I'll try to incorporate the fix in next release.

YosypC, 2021-09-20:
The problem seems to be resolved in one of the latest FC versions as [this](url=https://forum.freecadweb.org/viewtopic.php?f=3&t=54353&start=20#p534424) user claims that the refine works. The OCC is the same as mine (updating info as I forgot them).

realthunder, 2021-09-20:
That's a separate problem. The fillet/draft should work regardless of using refine or not. But yes, if that middle edge is removed by refine, then it will work. BTW, doing refine is kind of disruptive especially in the middle of a long history. It is recommended to only perform refine at the last step, or at least enable refine in only a few chosen one.