realthunder/FreeCAD_assembly3 #835: Add option to use Hold Right-Click and move mouse to rotate camera

ChaosBlades, 2021-09-17:
Currently the right click is only used to change navigation style. It has been extremely frustrating using FreeCAD when right click is not camera rotation since that is how I use Fusion 360, TinkercAD, Ultimaker Cura, and PrusaSlicer. All the modeling tools I use have the below camera movement hotkeys except FreeCAD so every time I try to use FreeCAD I have to unlearn muscle memory.

Right-Click to rotate
Middle Mouse button to pan
Scroll Wheel to zoom

herrgerd, 2021-09-18:
PrusaSlicer does not rotate the view with RMB, it pans. Fusion 360, Inventor and SolidWork don't either.
Haven't used Cura or TinkerCAD, so can't tell what these do..

ChaosBlades, 2021-09-18:
You are correct for PrusaSlicer. I just started using it and somehow didn't realize. Fusion 360 has an option to use TinkerCAD navigation. I set it so long ago I forgot about it. Inventor and SolidWorks are expensive so I have never used them.

OnShape also uses right click to rotate as default

Regardless I am used to it and holding a hot key to rotate when right click is doing nothing useful seems dumb.

bambukouk, 2021-09-18:
I use shift+right click in FreeCAD to rotate (with 3D naviagtion set to CAD mode), middle to pan, scroll wheel to zoom.
Which navigation mode are you using?
https://wiki.freecadweb.org/Manual:Navigating_in_the_3D_view

ChaosBlades, 2021-09-18:
I know how to rotate. The problem is none of the options let you set right-click to rotate. I am using the Blender setting but it is really frustrating. I'll probably just keep using Fusion 360 or move to OnShape until FreeCAD allows me to right click to rotate.

bambukouk, 2021-09-18:
If you look at FreeCAD forum you will see many examples of new users wanting FreeCAD to behave exactly in a way that is familiar to them from some other CAD system 😉

ChaosBlades, 2021-09-18:
If FreeCAD is not interested in making it as easy as possible for people coming from other CAD software to use it then I guess I have lost faith in the project ever improving to become a true replacement for the other offerings.

- RMB currently does nothing except open a menu to change navigate style. Why we need a hotkey to do this is beyond me.
- This would be fairly trivial to implement since it is a simple rebinding of keys which FreeCAD is already offering.

I have made my case for this to be implemented. It either does get implemented and I will use FeeCAD or I guess I am going to use OnShape so I can finally ditch Windows for Linux. Being forced to use a 2 button hotkey to perform an action I use constantly is not something I am interested in. Other CAD software that don't offer a RMB to rotate have a tightly integrated shortcut menu dedicated for that key. Which in that case an argument can be made but in FreeCADs case unless I have missed something it does not have a shortcut menu dedicated to RMB or at least one that is enabled by default.

herrgerd, 2021-09-19:
> If FreeCAD is not interested in making it as easy as possible for people coming from other CAD software to use it then I guess I have lost faith in the project ever improving to become a true [...]

With that attitude (like small children: I want that or I'll bwaaaaaah for hours) you won't move much. Not here, not in life.
I guess @realthunder is busy programming, so I'll tell you instead:

This branch here is done by _one person_. He's made FreeCAD so much better and more usable over the years, more than the original creators. So please, show some love for his work instead of snowflakin' around...
No-one forces you to use FreeCAD.

ChaosBlades, 2021-09-19:
I am not demanding anything. I put in a request. I am not even asking for it any time soon. If there is no time for this that is fine. But that is not what I got. This thread is the same reason this fork exists. The community puts in request, bug reports, merge request and they get nothing but attitude back.

realthunder, 2021-09-19:
@ChaosBlades There is mostly just me doing programming  in this fork. And as you can see, there are lots of issues pending. 

The feature you want, or more broadly speaking, a user customizable navigation style, is certainly good to have. But I can assure you, this is a big feature, and given the current architecture, a rather challenging one. I remember someone mentioned in Twitter wanting to work on this in upstream. So who knows, maybe you can get it earlier than you think.

ChaosBlades, 2021-09-19:
If you run across the twitter post again or an 'official' ticket on the main branch issue tracker post it here so I and others can follow the progress. I searched around on the main branch tracker and I don't see anything. The only forum post I can find is just replying to the request with the current navigation styles wiki. Seems the last navigation style to get added was blender back in 2016. Which is the most usable one for me.

Sure would be nice if main branch would move to a github issue tracker instead of that early 2000's looking thing.

luzpaz, 2022-01-19:
Have you tried middle mouse click + right click ?
Source: https://twitter.com/Chris_Gammell/status/1483460813629370370