realthunder/FreeCAD_assembly3 #836: Add Export as 3MF

ChaosBlades, 2021-09-17:
I am not seeing any option to export as a 3MF format.
The industry is moving away from STL and to 3MF. FreeCAD should support this.

https://en.wikipedia.org/wiki/3D_Manufacturing_Format

Fusion 360, Cura, and PrusaSlicer all support 3MF

herrgerd, 2021-09-18:
You can adjust the quality of .STL output:
[image]

You can use .amf too, at least as long as .3mf isn't supported yet.

herrgerd, 2021-09-19:
@realthunder just curious: Is the addition of another export-format in your real or is it something the creators of the main branch would do?

I read up on it and it seems like .3mf is completely free and open source: https://3mf.io/specification/

Cheers, Jan

ChaosBlades, 2021-09-19:
This is all I found from the main branch. It seems there has been little movement since Nov 2018. There is nothing in the bug tracker. FAQ wants you to start a thread before posting feature requests to the tracker.
https://forum.freecadweb.org/viewtopic.php?f=10&t=32165&start=20

This is the most up to date comparison I could find. AMF seems comparable to 3MF but 3MF has more compatibility.
https://xometry.eu/en/3d-printing-file-formats-compared-obj-stl-amf-and-3mf/

herrgerd, 2021-09-19:
Prusa has a good article on it, too: https://blog.prusaprinters.org/3mf-file-format-and-why-its-great_30986/
To be fair, I never encountered any of the "problems" mentioned with standard STL or AMF files. Of course the file format in itself is flawed and I absolutely hate it.

Slicers should import STEP files like every other program in the industry does. But as this ain't gonna happen we need to get along with imperfect triangulated files...

kevenwyld, 2021-09-20:
> You can adjust the quality of .STL output:
> [image]

Off topic, but did you do something special to get that `Mesh Formats` tab to appear?  I'm in the latest Link Daily appimage and don't see that.

ChaosBlades, 2021-09-20:
I can confirm on AppImage you need the mesh design work bench active. That is a little convoluted. Probably why I could not find it originally. I wonder what other settings only appear when you select a specific work bench.

luzpaz, 2021-09-21:
@ChaosBlades originally it was about memory ccnsumption, IIRC. Boot in to FC and load all the workbenches would be immensely resource draining. There may have been some technical difficulties also that led to this. The downside is users not being able to find the preferences they're looking for. In upstream there is a solution now coded by 'chennes' that lets the user specify what workbenches they want to default load at startup. I'm not sure it's it's in RT's Link branch. 

[image]

ref: https://forum.freecadweb.org/viewtopic.php?f=8&t=60074&start=30#p523481

ShivamJoker, 2021-09-23:
So @realthunder any plans for adding this ? We would love to have it

luzpaz, 2021-09-23:
@ShivamJoker that is no small feat, I imagine. Maybe let's let realthunder get toponaming sorted out before we ask for more features?

HaasNL2, 2021-09-27:
I don't feel like 3fm is real important for Freecad, it is more for 3D printers and with the use of the Slicer export plugin it really is not needed to safe it as 3FM

ShivamJoker, 2021-09-27:
Is there any difference in quality of amf and 3mf ?

ChaosBlades, 2021-09-27:
It is definitely not "real important" but it is a feature that a lot of CAD software has that FreeCAD does not. Fusion 360 for example exports 3MF by default now. I seen that no issue was raised for it on this branch and seeing as the main branch has not made any progress it makes sense to at the very least open an issue to track progress or lack there of. Having this mentality that FreeCAD needs / does not need X feature is not doing it any good. What FreeCAD needs is to become more inline with the features of other CAD software and other CAD software supports 3MF and FreeCAD does not. With that said this should definitely should be tagged as 'nice to have' / 'low priority' but saying FreeCAD does not need it outright is like I said not doing it any favors.

ChaosBlades, 2021-09-27:
This is interesting. Seems Cura is looking to enable importing of Step files and explains why they like 3MF.
https://community.ultimaker.com/topic/36279-import-step-files-in-cura/

HaasNL2, 2021-09-27:
PRusa has a better explanation 
https://blog.prusaprinters.org/3mf-file-format-and-why-its-great_30986/
But then again Freecad default format is better, because you won't lose any information and is still open source like 3mf.
Should ask slicers if they can support freecad format.

luzpaz, 2022-01-19:
> Should ask slicers if they can support freecad format.   

This!

eduncan911, 2022-01-24:
CURA supports importing FreeCAD.  Really wish Slic3r/PrusaSlicer/SuperSlicer would add FreeCAD.

luzpaz, 2022-01-24:
@eduncan911 do you know of any efforts to get Slic3r/PrusaSlicer/SuperSlicer to add FC ?

eduncan911, 2022-01-24:
@luzpaz nope, zero.

eduncan911, 2022-01-24:
Created a Feature Request, so we'll see how that goes: 

https://github.com/prusa3d/PrusaSlicer/issues/7808

ShivamJoker, 2022-01-25:
> CURA supports importing FreeCAD.  Really wish Slic3r/PrusaSlicer/SuperSlicer would add FreeCAD.

It keeps giving me error of unsupported file format.

Any mor details on this?

luzpaz, 2022-01-25:
anyone want to open tickets for Slic3r and SuperSlicer ?

eduncan911, 2022-01-25:
> > CURA supports importing FreeCAD.  Really wish Slic3r/PrusaSlicer/SuperSlicer would add FreeCAD.
> 
> It keeps giving me error of unsupported file format.
> 
> Any mor details on this?

You need to have FreeCAD installed locally, and it's CLI binary available in the PATH env variables that CURA is launched with.

https://community.ultimaker.com/topic/23548-freecad-plug-in/

kevenwyld, 2022-01-25:
So, I realize that freecad format in slicers would be great but in my opinion this does not invalidate including 3mf export capability, the original topic of this issue.  There's something to be said for conforming to "standards" even if better options are included as well.

That said, I'm not sure that the link branch is the right place for this.  And I just poked around and found that maybe 3mf support is already included in upstream master??? not sure as I haven't tried it yet:

https://github.com/FreeCAD/FreeCAD/commit/e3ebe4bc9f

luzpaz, 2022-01-27:
@kevenwyld relevant forum threads:. 
* https://forum.freecadweb.org/viewtopic.php?style=3&f=8&t=63095
* https://forum.freecadweb.org/viewtopic.php?t=32165

I think we should submit 3mf related commits upstream so they trickle in to Rats Link branch

eduncan911, 2022-01-27:
Looks like it's already added to FreeCAD core!

https://github.com/FreeCAD/FreeCAD/commit/e3ebe4bc9f

So, at this point, we just wait for the up-n-coming 1.20 release and realthunder can merge the upstream changes at some point.

Close this issue?

eduncan911, 2022-01-27:
> > > > CURA supports importing FreeCAD.  Really wish Slic3r/PrusaSlicer/SuperSlicer would add FreeCAD.
> > > 
> > > 
> > > It keeps giving me error of unsupported file format.
> > > Any mor details on this?
> > 
> > 
> > You need to have FreeCAD installed locally (no app image), and it's CLI binary available in the PATH env variables that CURA is launched with.
> > https://community.ultimaker.com/topic/23548-freecad-plug-in/
> 
> Where do I find its CLI binary?

Depends if you are Windows, macOS, or Linux.  But, should be in you PATH regardless of OS.  The question is, are you launching CURA with the same environment variables, that includes your PATH.

Open a command prompt and try to type: `FreeCADcmd`

If not found, then your PATH needs to include your FreeCAD install location for binaries.

Luberry, 2022-03-28:
> Looks like it's already added to FreeCAD core!
> 
> [FreeCAD/FreeCAD@e3ebe4bc9f](https://github.com/FreeCAD/FreeCAD/commit/e3ebe4bc9f)
> 
> So, at this point, we just wait for the up-n-coming 1.20 release and realthunder can merge the upstream changes at some point.
> 
> Close this issue?

Looks like you have to convert to a mesh first in the mesh wb, ideally you would be able to just be able to export using std export. though, a macro could probable be created to create a temporary mesh, export, remove the mesh. or just add it to std export