realthunder/FreeCAD_assembly3 #840: [issue] Fillet gives an error (BRep_API: command not done)

anao00, 2021-09-19:
Steps to reproduce:
1. Use FreeCAD-asm3-Daily-Conda-Py3-Qt5-20210916-glibc2.12-x86_64.AppImage
2. Open the .tar.gz 'fillet_error.FCStd' file attached
3. Recompute the 'FilletThatFails' object
4. Compare to the 'FilletThatWorks' object

Screenshot of the error:
[image]

As is seen, the Fillet and Chamfer tools work only selectively depending on the edge highlighted.
This error does not occur on' FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage' from the regular version of FreeCAD. However, if the fillet is applied successfully on the regular version, the file saved, and then opened on the LinkStage3 version, the recompute operation requested fails. 

Operating system: GNU/Linux 5.11.0-34-generic (Ubuntu 20.04)
FreeCAD AppImage hash: 4e48e6967af6b2c384066c9958f1b95e85bb1e38 (sha1sum)

[fillet_error.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/7192109/fillet_error.tar.gz)

realthunder, 2021-09-21:
Are you sure you are using `FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage`? Because I used the exact same version, and it fails as well, but with a different message `Resulting shape is invalid`. The `Command not done` message is probably related to #834.

I modified the code a bit in the referenced commit to show the invalid shape. See the highlighted defects. If you look closer, you can see the original geometry have some irregularity around that area, which causes OCCT fillet to fail.

[image]
[image]

anao00, 2021-09-21:
>Are you sure you are using FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage? 

It worked with FreeCAD_0.19-24291-Linux-Conda_glibc2.12-x86_64.AppImage on another file with the same solid.
I'm sorry, I never tested it with the new file because I assumed it would produce the same result.

>If you look closer, you can see the original geometry have some irregularity around that area, which causes OCCT fillet to fail.

Is there a solution for that that could be implemented linkstage3 or is it one of those chronic problems that won't likely be solved, ever?