realthunder/FreeCAD_assembly3 #842: pockets become pads when changing length of pad

kevenwyld, 2021-09-19:
I recall seeing something similar in a previous bug but I can't find it now.  Please close if it's a duplicate.

I have a part, which has pads in multiple directions from a plane.  The geometry for this sketch is derived from portions of a master sketch by selecting individual wires in the 3d view and clicking pad (a feature I really like by the way.)  This automatically creates a binder for those wires and uses them to define a pad.

It appears that when I go back to a pad near the beginning of the tree and try to change it's length though everything kindof explodes.  In this example all the pockets I have on the part turn in to pads, resulting an a completely broken part.  No errors are produced in the console though.  Returning the pad to the original value does not fix the problem, but clicking undo does.

I tested in v0.20 vanilla freecad and the model breaks, but in a more useful way.  An error is produced by the `clamp-holes-clearance-sketch` object reporting `AttachEngine3D: subshape not found`

Here's a video of the issue:

https://user-images.githubusercontent.com/11936535/133944295-42e75fc6-1702-4383-b34d-978409c8686a.mp4

And here's the file
[motor_mouting_bracket.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7192988/motor_mouting_bracket.zip)

Here's my system info:

```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.916.24301 +4152 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 7fd36f96f4db17db4f8d058eab22588cca8f9d5c
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

kevenwyld, 2021-09-21:
It seems that what's really happening here is the pockets are mysteriously switching from the `Subtractive` operation to the `Additive` operation.  I had this issue again on a much simpler part and opened the pocket parameters to find `Additive` was selected.  Switching to the appropriate operation resolved the issue.

realthunder, 2021-09-21:
Did you happen to author the file, or at least open and saved this file using 0911 release? There is a known bug in 0911 release that causes the subtractive feature becoming additive. If so, you'll have to manually correct it by select all the subtractive features, and set their `AddSubType` property to `Subtractive`. You can easily spot the wrong type by its icon. If it has 'Yellow' color than it is additive. Having red means its subtractive.

kevenwyld, 2021-09-21:
No I did not use the 9/11 release at all as I saw your message about the issues with that release and waited for the next one.  All these files, including the two I was having issues with today, were authored in the 9/16 one.

I keep automated versioned backups of all files I edit in Freecad, especially these development builds, so no work was lost.

I've attached a video and a file of a quick test I just did in the 9/16 appimage release.  Here's the version info again.
```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.916.24301 +4152 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 7fd36f96f4db17db4f8d058eab22588cca8f9d5c
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

Video: 

https://user-images.githubusercontent.com/11936535/134107586-6e68a3ed-da69-48fe-acd2-3c525748897f.mp4

File:
[test_subtractive_bug.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7200021/test_subtractive_bug.zip)

Bear with me on the video, it was difficult to force a recompute of that particular object but as you can see it says it's an additive operation as soon as I save the pocket.  After reopening the file or triggering a recompute the 3d view updates accordingly.

Thanks!

realthunder, 2021-09-21:
OMG! Just found out that the 0916 linux build didn't update its source code. It is still the 0911 version. Now I am in big trouble.

kevenwyld, 2021-09-21:
uh oh, sorry to hear that.  Let me know if I can help somehow.

kevenwyld, 2021-09-22:
Everything looks good in the 9/22 release.  My pockets are staying subtractive in all the models I've tested.  Thanks!

bambukouk, 2021-09-25:
I am also using 9/22 and all is fine, 
but "pockets are staying subtractive" is not quite as it is - you have  options and you have to choose which one you want:

> New Shape

> Additive

> Subtractive`

> Intersecting

I like it :+1:  this is great !

kevenwyld, 2021-09-25:
> but "pockets are staying subtractive" is not quite as it is

@bambukouk yeah what I meant by that was in the previous release if you selected subtractive in the pocket dialogue and clicked OK it would silently change to additive.  That bug appears to have been resolved.

> when I choose to edit existing pockets (created in earlier versions of FreeCAD) - they don't necessarily default to subtractive

I have not had any issues opening old models as long as they were created in a release older than the 9/11 one.  Can you provide an example?  Maybe there is still an issue related to this change?

bambukouk, 2021-09-25:
> I have not had any issues opening old models as long as they were created in a release older than the 9/11 one. Can you provide an example? Maybe there is still an issue related to this change?

I don't have issues **"opening"** old models (they are fine)
I am talking about **"editing"** of old/existing pockets.
These are, of course, two different things.

Historically they would simply be subtractive (no other options).

What I am saying is that now if you open existing/old pocket (to edit it), as part of editing you will have to specify which option you want (even though it was originally created as subtractive).

Not an issue, really, just a "got you" 

Hope that this clarifies what I was trying to say :+1:

kevenwyld, 2021-09-25:
Interesting.  I just tried modifying some pockets in a few older medium sized assemblies/parts.  Subtractive is the default for pockets, even those created in earlier versions, which seems correct.  I imagine if I kept digging I'd find what you are referring to, but I agree it's probably not a problem.  I believe in general compatibility with older models isn't the main focus of this fork.  It's nice that it's been mostly OK so far though.

All my fillets break though if I click OK in the dialogue, which is to be expected :grin:

bambukouk, 2021-09-25:
Yes, I am so delighted about this enhancement :+1:

closed 2021-09-25 by kevenwyld