realthunder/FreeCAD_assembly3 #843: [BUG: part design] Pad/pocket not working with base feature

Miksu80, 2021-09-20:
Basically pads and pockets "forgets" the base feature totally :(

https://user-images.githubusercontent.com/48354615/133969660-3cc3e442-49d0-43b3-bad8-05a987f50901.mp4

File:
[LaserMount_edit.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7194226/LaserMount_edit.zip)

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.916.24301 +4152 (Git)
Build type: Release
Branch: LinkDaily
Hash: 7fd36f96f4db17db4f8d058eab22588cca8f9d5c
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

realthunder, 2021-09-21:
The problem is the base feature does not count a solid shape. It is a shell instead. The above commit will try to auto create the solid in PartDesign base feature.

closed 2021-09-21 by realthunder

Miksu80, 2021-09-21:
That's correct, my mistake, sorry for this. 
Would it be possible to adjust for example edge line color to a different one when having shells? There (at least to my knowledge) is no direct way to know if its solid or not but you've got to cut the view.