realthunder/FreeCAD_assembly3 #848: [BUG: Part] Mirroring shape produces square shapes instead of round ones

Miksu80, 2021-09-23:
So basically, made a design in part design WB, used Lattice also if matters and tried to make a simple mirror using PART WB mirror feature and it failed (I need both of them and it is not possible make in part design so that both sides would be preserved)

[Mirror_issues.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7215477/Mirror_issues.zip)

[image]

Got around it and working in "Body002" by using base feature

Also the Lattice planes dont have surface colors on "as is" -draw style so its not so easy to see the top or bottom of the plane

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

mwlandolsi, 2021-09-23:
Hello,
[image]
I have re-designed this part but with my own workflow.
could this be useful for you ?
[test part symetry.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7216110/test.part.symetry.FCStd.zip)

Miksu80, 2021-09-23:
Thank you 👍, 
I got it already working in the original (i've hidden the Body002 which is working). In the model you uploaded, all the latches are the same way (all facing same direction) so it wouldn't work on my case. I would have done the latch fillets in lattice base model but it fails so it didn't help me as much as i'd liked 😶

bambukouk, 2021-09-23:
Why use Part mirror, when link with -1 scale is so much better?
try it 👍

Miksu80, 2021-09-24:
I used mirror just to point out this bug 🤥
I'll try to remember the -1 link scale, thanks 👍

bambukouk, 2021-09-24:
I have tried openning both files ("Mirror_issues.zip" and "test part symmetry.FCSTD.zip") and for some reason the message I am getting is:

> 16:53:27  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/tmp/.mount_FreeCASBaJQd/usr/Mod/Arch/importSH3D.py", line 47, in open
    read(filename)
  File "/tmp/.mount_FreeCASBaJQd/usr/Mod/Arch/importSH3D.py", line 79, in read
    homexml = z.read("Home.xml")
  File "/tmp/.mount_FreeCASBaJQd/usr/lib/python3.9/zipfile.py", line 1463, in read
    with self.open(name, "r", pwd) as fp:
  File "/tmp/.mount_FreeCASBaJQd/usr/lib/python3.9/zipfile.py", line 1502, in open
    zinfo = self.getinfo(name)
  File "/tmp/.mount_FreeCASBaJQd/usr/lib/python3.9/zipfile.py", line 1429, in getinfo
    raise KeyError(
<class 'KeyError'>: "There is no item named 'Home.xml' in the archive"

Strange... (other files are openning normally)

`OS: elementary OS 5.1.7 Hera (Pantheon/pantheon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United Kingdom (en_GB)
`