realthunder/FreeCAD_assembly3 #849: chamfer near a threaded hole causes parts of the thread to dissapear

kevenwyld, 2021-09-24:
This is a weird one.... When I create a chamfer on an edge which is near a threaded hole, the outside faces of the threaded hole disappear.  I captured a video of the problem below.  Looking at the video you can see that the outer vertical edge of the thread disappears after clicking OK on the create chamfer dialogue.  I only noticed this when I went to 3d print a threaded part and the slicer couldn't make sense of the broken faces.  It's very hard to see in FreeCAD.

Video:

https://user-images.githubusercontent.com/11936535/134703808-634bda21-c7d2-4134-9a18-3234927347ef.mp4

File with chamfers and broken faces:
[chamfer-threaded-hole-bug.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7226935/chamfer-threaded-hole-bug.FCStd.zip)

kevenwyld, 2021-09-24:
So, after further investigation, this seems to be an issue if I add a chamfer or fillet on any part of a body containing modeled threads.

kevenwyld, 2021-09-24:
So I have not been able to recreate this issue after restarting FreeCAD.  The old files are still broken when I open them, including the one I posted in this issue, but creating a new document with a threaded hole now behaves correctly.  I have no idea what happened.

realthunder, 2021-10-12:
The missing faces is caused by a bug in the `Refine` operation. You can work around it by turning off `Refine`. The above reference commit has fixed the bug.