realthunder/FreeCAD_assembly3 #853: Cannot constrain assembly of assemblies to another one

not-a-number-dot-org, 2021-09-25:
Hi

So I have in one file this table which is made of something like 125 parts (but maybe only 20/25 differents one) which are assemblies containing only bodies. They are assembled using 99% of Plane Alignment constraints.

And on another file I have the drawers which is an assembly of 6 assemblies (stored in various files) that I'am trying to assemble with the table.

But I cant because I have an error on the first constraint that I try to set (see last image below) (I tried with different elements (faces))

Not sure if I was supposed to post this on FCAD forum instead.
btw is there an option to have more verbose error messages ?

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: French/France (fr_FR)

[image]
[image]
[image]

realthunder, 2021-09-25:
According to the error, your assembly is too big. The solver has a hard coded limit of maximum 1024 parameters to solve for a single assembly. Each part has six unknowns corresponding to its six DOF. I'll probably raise that limit in future release. But still, it is recommended to break large assembly into smaller sub-assemblies. Just like doing real assembling, it's good to modularize things.

not-a-number-dot-org, 2021-10-17:
Hello.

Any chance to have an update with a dynamic parameter limit soon ?
My table frame is just missing 24 constraints to be finished.

Thanks in advance.

closed 2022-01-05 by realthunder