realthunder/FreeCAD_assembly3 #855: Path Sanity

snelweg, 2021-09-26:
There is something wrong with the Path workbench in LinkStage3 I guess.
I made a very simple file that works with the Path Sanity command in the official FC version 0.19 (appimage). My home compiled version of LinkStage3 doesn't work and gives errors.

Error message in LinkStage3:
```

PathJobGui.INFO: Expected a specific object to edit - Job not recognized
Running the Python command 'Path_Sanity' failed:
Traceback (most recent call last):
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathSanity.py", line 119, in Activated
    html = self.__report(data)
  File "/home/ton/FREECAD/daily/freecad/Mod/Path/PathScripts/PathSanity.py", line 407, in __report
    reportraw = self.outputpath + job.PostProcessorOutputFile + '.asciidoc'

```

It works in :
```
 OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)
```

LinkStage3 version:
```

OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28504 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: dbf51658a4f8f175dba82d350b3fc0a96ae7c332
Python version: 3.9.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)

```
I started both versions with a clean .Freecad directory. I tried the LinkStage3 version with the .Freecad dir from the official appimage and that didn't solve the problem either.
Would be real nice if this would work, I can import the Sanity html files into LinuxCNC so the operator can see what the steps for fabrication are.

snelweg, 2021-09-27:
There seems to be a problem with the 'job' as it is in LinkStage3. There is a difference between the way 'official' and the superior version handle the job object. 
Maybe a good time now to upgrade the Path workbench to 0.20?

realthunder, 2021-09-27:
On the contrary, I have merged the `Path` workbench with the upstream as recent as about a month ago. There is indeed quite a lot of changes to the `Path` workbench since 0.19. Please try it with official 0.20 first. I will perform a full merge with upstream in the next release or two.

snelweg, 2021-09-27:
> On the contrary, I have merged the `Path` workbench with the upstream as recent as about a month ago. 

That is great to hear!

>Please try it with official 0.20 first. 

I will do that right now.

>I will perform a full merge with upstream in the next release or two.

Wow, that would be awesome! Your version is such fun to work with, and it doesn't crash anymore like it used too (as did the official one btw).
Thanks!

snelweg, 2021-09-27:
You were right ofcourse.
Exactly the same problems in official 0.20

```
10:10:09  PathJobGui.INFO: Expected a specific object to edit - Job not recognized
10:10:09  Running the Python command 'Path_Sanity' failed:
Traceback (most recent call last):
  File "/tmp/.mount_FreeCAfQTni0/usr/Mod/Path/PathScripts/PathSanity.py", line 119, in Activated
    html = self.__report(data)
  File "/tmp/.mount_FreeCAfQTni0/usr/Mod/Path/PathScripts/PathSanity.py", line 407, in __report
    reportraw = self.outputpath + job.PostProcessorOutputFile + '.asciidoc'

name 'job' is not defined
```

```
OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/lightdm-xsession)
Word size of FreeCAD: 64-bit
Version: 0.20.25844 (Git) AppImage
Build type: Release
Branch: master
Hash: 1b20118dfa3fb9f127f0e4fb122701f7fcdaa9fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

```

```

Python 3.9.7 | packaged by conda-forge | (default, Sep 23 2021, 07:39:18) 
[GCC 9.4.0] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',2)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> ### Begin command Std_RecentFiles
>>> FreeCAD.openDocument('/home/ton/testSanityAgain.FCStd')
>>> # App.setActiveDocument("testSanityAgain")
>>> # App.ActiveDocument=App.getDocument("testSanityAgain")
>>> # Gui.ActiveDocument=Gui.getDocument("testSanityAgain")
>>> ### End command Std_RecentFiles
>>> Gui.runCommand('Std_PerspectiveCamera',1)
>>> # Gui.Selection.addSelection('testSanityAgain','Job')
>>> Gui.runCommand('Path_Sanity',0)
>>> # Gui.Selection.clearSelection()
>>> Gui.runCommand('Std_About',0)
```

luzpaz, 2022-04-13:
@snelweg would you be so kind to report this to Path [subforum](https://forum.freecadweb.org/viewforum.php?f=15)?

snelweg, 2022-04-14:
I will. Problem still here in version daily 2407

`OS: Debian GNU/Linux 11 (bullseye) (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2022.407.28576 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 599f7663af69d3dbd77063e834b12f26da513b2d
Python version: 3.9.12
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: Dutch/Netherlands (nl_NL)
`

closed 2022-04-14 by snelweg