realthunder/FreeCAD_assembly3 #856: clicking subtractive loft defaults to additive operation

kevenwyld, 2021-09-27:
Clicking the subtractive loft button in part design defaults to additive operation.

Video:

https://user-images.githubusercontent.com/11936535/134842094-d7d2046a-0914-4ad2-9400-437569133633.mp4

snelweg, 2021-09-27:
Maybe related:

https://www.patreon.com/posts/56457476

kevenwyld, 2021-09-27:
@snelweg thanks, though the screen capture above was in a new document for the purpose of demonstrating this bug, and I'm on the 9/22 release.

I didn't initially upload a file because you have to actually click loft to recreate the bug, but here's the base setup I used for the screen capture.  Select Sketch001 followed by Sketch002 and click `Subtractive Loft` from the toolbar.  For me the dialogue opens a loft where the operation is Additive.

File:
[additive_subtractive_loft_bug.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7237432/additive_subtractive_loft_bug.FCStd.zip)

closed 2021-09-28 by realthunder