realthunder/FreeCAD_assembly3 #857: mirrored subtractive loft results in broken geometry

kevenwyld, 2021-09-27:
I have a part which I'd like to put little handle recesses on.  I created one with a subtractive loft and attempted to mirror it to the other side however the mirror appears to break the part.

Video:

https://user-images.githubusercontent.com/11936535/134842762-fcd00006-5c14-4ecf-afbc-404cb59cdd86.mp4

File:

[case_assembly.FCStd.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7233222/case_assembly.FCStd.zip)

```
OS: Arch Linux (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)
```

kevenwyld, 2021-09-28:
This seems to have more to do with the boolean that was earlier in the tree.  I can't figure it out though, it seems to only be in this file (created in the 9/22 release).

I attempted recreating my steps in another file and though time did not permit getting things exactly the way they were, the basic setup (assembly containing body with references to another body outside the assembly) did not recreate the problem.  But still something remains broken in this model and no errors are reported.

It seems like the binder that the boolean operation creates to the original shape, which is itself binder, looses some of it's edges.  See the included screen captures for more detail.

https://user-images.githubusercontent.com/11936535/135016290-6625c786-206f-4fd7-bb23-3be6bcb04082.mp4

https://user-images.githubusercontent.com/11936535/135016399-c59f48f7-2dd0-477c-bbab-59499eafd70f.mp4

kevenwyld, 2021-09-28:
Also I'm not sure if it's helpful but I ran `Part->Check Geometry` on the binder that appears broken (inside the boolean object.  Here are the results which show several issues and consider it an invalid shape.

[image]

Tagging a couple of the names of the references here for searching:
```
Vertex3 Vertex BOPAlgo SelfIntersect
Edge13  Edge   BOPAlgo TooSmallEdge
```

kevenwyld, 2021-09-29:
Well I fixed it by recomputing absolutely everything in the entire assembly (select all in the tree -> mark to recompute -> recompute).  I'm not sure why this was necessary though.  I'll leave this here for a few days and then close it if no one comments.

kevenwyld, 2021-09-29:
#823 looks shockingly similar to this issue.  Recomputes required when they shouldn't be and strange mutated geometry.

realthunder, 2021-10-10:
> #823 looks shockingly similar to this issue.

Yes indeed, and also #854. A work around fix has been added recently.