realthunder/FreeCAD_assembly3 #860: Arch & BIM

eustachi0, 2021-09-29:
Levels and WPProxy not working. 
Apologies in advance if it's me doing something wrong. I am new to FreeCAD.

**Level (Floor) - Right Click - Set working plane**
```
23:22:02  Pick a face, 3 vertices or a WP Proxy to define the drawing plane
23:22:02  Running the Python command 'Draft_SelectPlane' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\draftguitools\gui_selectplane.py", line 134, in Activated
    if FreeCAD.DraftWorkingPlane.alignToSelection():
  File "C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\WorkingPlane.py", line 658, in alignToSelection
    return self.alignToFace(shape, offset) \
  File "C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\WorkingPlane.py", line 550, in alignToFace
    if shape.ShapeType == 'Face':

cannot determine type of null shape
```

**WPProxy - Double Click**
```
Pick a face, 3 vertices or a WP Proxy to define the drawing plane
23:24:36  Running the Python command 'Draft_SelectPlane' failed:
Traceback (most recent call last):
  File "C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\draftguitools\gui_selectplane.py", line 134, in Activated
    if FreeCAD.DraftWorkingPlane.alignToSelection():
  File "C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\WorkingPlane.py", line 658, in alignToSelection
    return self.alignToFace(shape, offset) \
  File "C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\WorkingPlane.py", line 550, in alignToFace
    if shape.ShapeType == 'Face':

cannot determine type of null shape
```

realthunder, 2021-10-03:
The above fix is just for the error you posted above. I am also not familiar with Draft workbench. If you have further problem, feel free to reopen or create another issue.

closed 2021-10-03 by realthunder

eustachi0, 2021-10-03:
Thanks. I guess I just need to download the daily build again and run it?

realthunder, 2021-10-03:
Not yet. I haven't made the new release yet. But since, this is only python code change. You can try to find the file `C:\Program Files\FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922\Mod\Draft\WorkingPlane.py`, and change line 550, from

```
if shape.ShapeType == 'Face':
```

to

```
if not shape.isNull() and shape.ShapeType == 'Face':
```

Save and restart FreeCAD.

eustachi0, 2021-10-03:
Thank you, I will try that and let you know.

eustachi0, 2021-10-03:
I have made the change you suggested, and although the error is gone the normal functioning of the BIM Level and WPProxy tools are not correct. I am using FreeCAD 0.20 for the examples showing the correct functioning (it works similar on FreeCAD 0.19.2 which is the version I am mostly using as I find it more stable for the Arch and BIM workbenches).
This is how BIM Levels should work:
[image]

This is how BIM WPProxy should work:
[image]

This is what happens using FreeCAD-asm3-Daily-Win64-Py3-Qt5-20210922:

[image]
As you can see, it does not set working plane and it does the same when I try to set a WPProxy.

No rush with this.
I am not working on a big project at the moment, I am in fact new to FreeCAD and I can see that I can start replacing my current workflow to use FreeCAD more professionally. I like the UI improvements you have implemented so far, especially the shadow view, clipping planes with fill and the show always on top when selecting an element with the transparency. Is the a way to bring these 3 features to FreeCAD 0.19.2?

Cheers,
Eustachio