realthunder/FreeCAD_assembly3 #861: [BUG:Sketcher] External geometry forms into wrong height

Miksu80, 2021-09-30:
Took an external reference and it appeared (correct shape) in wrong height:
[image]

File:
[External geometry fails.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7258778/External.geometry.fails.zip)

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Finland (en_FI)

Miksu80, 2021-09-30:
This might have something to do with this:

https://user-images.githubusercontent.com/48354615/135460750-e7a5feab-020b-46d7-814e-bd23b2347201.mp4

It looks like the sketch horizontal center line is off.

File:
[Dim_offset.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7259707/Dim_offset.zip)

realthunder, 2021-10-02:
This looks like a problem from upstream. But since there is no defining external geometry in upstream, it doesn't show up. It only happens to external bspline.

closed 2021-10-02 by realthunder

realthunder, 2021-10-02:
You second post is about a different problem, which arguably is not a bug. The interpretation of symmetric is dynamic because the `Sketcher` tries to be user friendly by guessing the intention. And because your sketch is ill formed, the guess is not good. Just delete the redundant constraint and retry the symmetric constraint. Even then the guess may not always be on spot. Just remember, by right, you are supposed to select the two symmetric points first, and the reference line/point last.

Miksu80, 2021-10-05:
Hi,

https://user-images.githubusercontent.com/48354615/135998979-ab60f744-cf94-4f94-b73b-8c6abdec815f.mp4

@realthunder do you have an idea how to solve this? I'm on newest version and its on all new parts -it is really bugging me :(

Regards, Mika

realthunder, 2021-10-05:
I don't understand your problem here? The screencast shows you've created a conflict constraint, hence the error when you trying to edit. That's expected, no? What do you mean by "the sketch horizontal line has the effect every time"?

Miksu80, 2021-10-05:
The problem is that when I constrain the point to horizontal sketch center line, it jumps some mm upwards. So basically if I draw a 20mm vertical line and constrain it to horizontal sketch center line, it becomes for example 26.34mm long.

realthunder, 2021-10-05:
The problem is fixed by [this](https://github.com/realthunder/FreeCAD/commit/ad1b5068b07b27a42ad8d42e2b15e7a5ad05c3d2) commit. Next time please open a new issue for a new problem. Meanwhile, you can use horizontal/vertical distance to work around this problem.

Miksu80, 2021-10-05:
Yep, thanks, I kind of wanted to verify this is going to be fixed and thought that this was a part of the same issue. 👍