realthunder/FreeCAD_assembly3 #863: During initialization the error "module 'slvs' has no attribute 'System'" occurred in freecad.asm3

snelweg, 2021-10-03:
FreeCAD daily build:

```
cd ~/FREECAD/daily/freecad-source/
git pull
cd ../freecad
cmake   ../freecad-source -DBUILD_FEM=OFF -DBUILD_QT5=ON -DBUILD_PY_LIB=ON -DBUILD_CXX_LIB=ON -DUSE_PY_3=ON -Wno-dev -Wdeprecated-declarations
time cmake --build /home/ton/FREECAD/daily/freecad -j 8
```

installed Assembly3 as per instructions https://github.com/realthunder/FreeCAD_assembly3

installed SLVS https://github.com/realthunder/FreeCAD_assembly3/issues/363 
and per `python3 -m pip install py_slvs
`

```
During initialization the error "module 'slvs' has no attribute 'System'" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 158, in InitApplications
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/ton/FREECAD/daily/freecad/Mod/FreeCAD_assembly3/freecad/asm3/init_gui.py", line 10, in <module>
    from . import sys_slvs
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/ton/FREECAD/daily/freecad/Mod/FreeCAD_assembly3/freecad/asm3/sys_slvs.py", line 45, in <module>
    class _SystemSlvs(SystemExtension,slvs.System):
AttributeError: module 'slvs' has no attribute 'System'
--------------------------------------------------------------------------------
```

```
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28527 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 1ecef0ecd7699129807918d6cca70f308215b82f
Python version: 3.9.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)
```

realthunder, 2021-10-03:
Please type the follow commands in FreeCAD python console, and post the output here

```
import py_slvs
py_slvs.slvs
py_slvs._slvs
```

snelweg, 2021-10-03:
> ```
> import py_slvs
> py_slvs.slvs
> py_slvs._slvs
> ```
```

Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',16)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> import py_slvs
>>> py_slvs.slvs
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: module 'py_slvs' has no attribute 'slvs'
>>> py_slvs._slvs
Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: module 'py_slvs' has no attribute '_slvs'
>>> 
```

realthunder, 2021-10-03:
Please remove directory `~/FREECAD/daily/freecad/Mod/FreeCAD_assembly3/py3_slvs` and restart FreeCAD. Type in the following command. What's the output
```
import py_slvs
py_slvs
```

snelweg, 2021-10-03:
> Please remove directory `~/FREECAD/daily/freecad/Mod/FreeCAD_assembly3/py3_slvs` and restart FreeCAD. 

```
:~/FREECAD/daily/freecad/Mod/FreeCAD_assembly3$ ls -la
total 80
drwxr-xr-x  6 ton ton  4096 Oct  3 12:12 .
drwxr-xr-x 31 ton ton  4096 Oct  2 08:24 ..
drwxr-xr-x  4 ton ton  4096 Oct  2 08:24 freecad
drwxr-xr-x  8 ton ton  4096 Oct  3 09:15 .git
drwxr-xr-x  2 ton ton  4096 Oct  2 08:24 .github
-rw-r--r--  1 ton ton  1157 Oct  2 08:24 .gitignore
-rw-r--r--  1 ton ton     0 Oct  2 08:24 .gitmodules
-rw-r--r--  1 ton ton     0 Oct  2 08:24 __init__.py
-rw-r--r--  1 ton ton 35141 Oct  2 08:24 LICENSE
-rw-r--r--  1 ton ton    37 Oct  2 08:24 MANIFEST.in
-rw-r--r--  1 ton ton  5249 Oct  2 08:24 Readme.md
-rw-r--r--  1 ton ton   609 Oct  2 08:24 setup.py
drwxr-xr-x 11 ton ton  4096 Oct  2 08:30 slvs
```

> 
> ```
> import py_slvs
> py_slvs
> ```

```
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',16)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> import py_slvs
>>> py_slvs
<module 'py_slvs' from '/home/ton/.local/lib/python3.9/site-packages/py_slvs/__init__.py'>
>>> 
```

Error output in terminal window:

```
During initialization the error "module 'slvs' has no attribute 'System'" occurred in freecad.asm3
--------------------------------------------------------------------------------
Traceback (most recent call last):
  File "<string>", line 158, in InitApplications
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/ton/FREECAD/daily/freecad/Mod/FreeCAD_assembly3/freecad/asm3/init_gui.py", line 10, in <module>
    from . import sys_slvs
  File "/usr/lib/python3/dist-packages/shiboken2/files.dir/shibokensupport/__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "/home/ton/FREECAD/daily/freecad/Mod/FreeCAD_assembly3/freecad/asm3/sys_slvs.py", line 45, in <module>
    class _SystemSlvs(SystemExtension,slvs.System):
AttributeError: module 'slvs' has no attribute 'System'
--------------------------------------------------------------------------------
connect failed: No such file or directory

```

realthunder, 2021-10-03:
Try

```
import py_slvs
from py_slvs import slvs
from py_slvs import _slvs
slvs
_slvs
```

It seems that you have trouble import the native Python plug in `_slvs.so` for some reason.

If the above not working. Please open your system Python interpreter and entering the above commands and post the output too.

snelweg, 2021-10-03:
> Try
> 
> ```
> import py_slvs
> from py_slvs import slvs
> from py_slvs import _slvs
> slvs
> _slvs
> ```

```
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',17)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> import py_slvs
>>> from py_slvs import slvs
>>> from py_slvs import _slvs
>>> slvs
<module 'py_slvs.slvs' from '/home/ton/.local/lib/python3.9/site-packages/py_slvs/slvs.py'>
>>> _slvs
<module 'py_slvs._slvs' from '/home/ton/.local/lib/python3.9/site-packages/py_slvs/_slvs.so'>
>>> 
```

> 
> It seems that you have trouble import the native Python plug in `_slvs.so` for some reason.
> 
> If the above not working. Please open your system Python interpreter and entering the above commands and post the output too.

```
ton@DESKTOP-9FIBCI1:~$ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import py_slvs
>>> from py_slvs import slvs
>>> from py_slvs import _slvs
>>> slvs
<module 'py_slvs.slvs' from '/home/ton/.local/lib/python3.9/site-packages/py_slvs/slvs.py'>
>>> _slvs
<module 'py_slvs._slvs' from '/home/ton/.local/lib/python3.9/site-packages/py_slvs/_slvs.so'>
>>> 

```

It's a standard install of Debian 11, fresh install last week.
```
OS: Debian GNU/Linux 11 (bullseye) (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.28527 (Git)
Build type: Unknown
Branch: LinkDaily
Hash: 1ecef0ecd7699129807918d6cca70f308215b82f
Python version: 3.9.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)
```

realthunder, 2021-10-03:
Restart FreeCAD, and try the following command
```
import slvs
slvs
```

Maybe you have another `slvs` module installed?

snelweg, 2021-10-03:
> import slvs
> slvs

```
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> Gui.runCommand('Std_Workbench',17)
>>> Gui.runCommand('Std_ViewStatusBar',1)
>>> import slvs
>>> slvs
<module 'slvs' (namespace)>
>>> 
```

> Maybe you have another slvs module installed?

I do not think so. This machine is a new install with Debian 11. I tried the recommended way to get slvs installed.  But I am usualy wrong, I will try to deinstall the slvs I have now.

>installed SLVS #363
>and per python3 -m pip install py_slvs

snelweg, 2021-10-03:
And of course you were right again. Sorry to have wastes your time.

I did a `pip3 uninstall py_slvs` removed the `Mod/FreeCAD_assembly3/` directory. Then did a `pip3 install py_slvs` and a 'git clone git@github.com:realthunder/FreeCAD_assembly3.git ' in the Mod directory.
Started the daily FreeCAD and it works. Accept for the normal errors I even get in the 'official'  version.

closed 2021-10-03 by snelweg