realthunder/FreeCAD_assembly3 #865: Fails TO LOAD STEP FILE WHILE GRABCAD ONLINE READER CAN DO EASILY

anirudhnehra, 2021-10-04:
Really, its pointless to even consider this freecad as a usable engnieering tool when it cant do what an online reader of step. Here is the link to step file:https://drive.google.com/file/d/1rkjPUH4aPEhQMZN0M71iWt4Sfywc-QR3/view?usp=sharing

All your time is spent on beutifying stupid UI while the core fucntionality remaains cracked.

luzpaz, 2021-10-05:
Please refrain from ALL CAPSing ticket titles. Remember to always copy/paste your full [About](https://wiki.freecad.org/About) info.

realthunder, 2021-10-10:
It's an OCCT bug, and has a simple fix. I'll submit a PR to them later. Can I use your file as a sample when reporting the problem?

realthunder, 2021-10-10:
Just checked that OCCT has fixed this problem in 7.6, which hasn't been released yet. Looks like we'll have to patch it by ourselves somehow.

anirudhnehra, 2021-10-10:
Yes sure you can use it.
Also, could you fix that memory leak problem of freecad. Freecad doesn't release the memory occupied by it when we create a rather big array of a part and then delete the array.

Regards 
Anirudh Nehra

realthunder, 2021-10-11:
Are you using the upstream FreeCAD in the screen capture? Please try with my `Daily` branch, and use the recommended STEP export setting below. The highlighted settings are most relevant for large complex models. 

You probably also want to enable the experimental renderer for better rendering performance, `Tools -> Preset configurations -> Display settings`. Note that there is some deficiency when using the transform tools in experimental render. I have just fixed that. It will not cause too much slow down when transforming in future release.

I have tried 100 instances with around 4GB memory usage. After applied transformation to one of the segment, the memory usage went up to 7GB, which I have already explained is due to recomputation pushes old shapes to undo stack. But that's only one of the reasons. You can't really tell how much memory the application is actually using relying solely on some type of external resource monitor. The system memory allocator may keep the allocated pages even the application has freed it, and only truly release the page when the kernel demands it because other application requires it. To prove this, you can close the document and open it again, perform the same transformation operation, and you'll find that FreeCAD memory usage stays about the same.

[image]

anirudhnehra, 2021-10-13:
Hi, 
1-Yes I am using the daily branch of your ASM3 version.
2-I am using the experimental renderer and yes it slows down a bit on using transform tool.
3-Is there a way to disable undo/redo feature which might be causing increased memory usage.
4-Maybe the system monitor is reporting incorrect value but the observed fact is:
when the system monitor shows close to 100 percent RAM usage(for example my system has 32 GB RAM and at around 31 GB occupied), the computer starts to hang up and the screen freezes. Opening and using any other application is not possible.
5-Yes I am using the same settings as shown by you. It still crashes.
6-Also, creating the array operation( say 400 by 400) and not using transform tool if I remember correctly. 
the issue is I create the array and freecad takes more RAM. But when I close the "same document" but not freecad, the RAM usage is still the same(high). It should come down with the closure I guess.

I have one other query. Is it possible to embed a macro script in the model tree? For example I write a macro that calculates mass before applying a fillet. I need this macro to always run before fillet is recomputed so that the mass of a body without filleting is reported. Is it possible make macro as a feature in model tree?

Regards and Thanks
Anirudh Nehra

realthunder, 2021-10-16:
The key to keep memory usage in check is to not building any compound shape with array. Even if OCCT compound is relatively cheap, it still occupies memory, and much more noticeable with array of complex model. Please make sure you are using the STEP import setting I posted above which avoid using any compound. And with the new 10.15 release, when creating draft array, you have the option to disable `Build shape`. You can also do the same with existing array using the context menu (right click the array icon in the tree view, and `Disable build shape`). In case you want to move an array element (not draft array, but link array imported from STEP), right click the array, and select `Toggle array elements`.

[image]

About the macro, how do you intend to report the mass? In my branch, you can write expression to do pretty much anything macro can do.

anirudhnehra, 2021-10-16:
Hi,
Ok I will try your setting. One improvement in paath array tool could be pattern by chordal length(straight distance between consecutive elements om the curve) .this tool will be very helpful to build chains of cycles or excavator /cable tracks.

Another improvement could be pattern by reference. That means suppose I build array of coordinate systems on a path. Then I make a body and want body to be coincident in position and orientation with each corresponding member of array of CSYS. I could then use attachment constraint of ASM3 and locate first body with the seed member of CSYS array.
Then I just say array on the body and all bodies are created (equal to number of CSYS) and oriented automatically to match the coordinate system array .
This feature of reference array is available in every major cad software. It also helps to reduce memory footprint.
Please see this series https://youtu.be/-R9myIG5XjA
I think Creo Parametric has the best pattern tools worth taking a look at.

Regarding the macro, it could report anywhere. It could write to a CSV file or in report view or create a model tree feature. The main issue is fixing the timestamp at which this macro must be executed using CTRL+R. It must allow to be executed after a feature of choice in the model tree.
Also, one more improvement could be feature suppression by variable using a spreadsheet.(0 could be feature off and 1 feature on).
Though freecad is great, we could always learn something from other software.
Freecad has to become primarily a mechanical design tool and I hope it reaches the same capabilities as the big three(Creo, NX, Catia)

Regards and Thanks and keep up the good work.
Anirudh Nehra