realthunder/FreeCAD_assembly3 #872: "GG" subshape selection does not select correctly Wire subshapes

carlopav, 2021-10-10:
Not sure if I got right how it is supposed to work, but I expected the subelement wire selector to select all the edges that were included into the Wire:

[image]

Thinking a bit more while writing, it's also true that the Wire is not a geometry, so probably displaying the edges as selected can be misleading, as the user have no visual difference between having all the wire's edges selected and the wire selected... 
What do you think?

realthunder, 2021-10-10:
I don't get what you mean here. According to your screencast, the selector does select all edges of the wire, no?

realthunder, 2021-10-10:
Maybe you mean when you click the wire menu item, it didn't select anything? Hmmm... probably a bug, can you please post your about info. Are you using the latest release?

carlopav, 2021-10-11:
> 
> 
> Maybe you mean when you click the wire menu item, it didn't select anything? Hmmm... probably a bug, can you please post your about info. Are you using the latest release?

Yes, exactly. here you have the about info:
```
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Italian/Italy (it_IT)
```

Wait, It works, sorry for the noise... Yesterday evening I was playing with Part.removeShape and probably I corrupted the wire... sorry for not having investigated more before posting the issue....

closed 2021-10-11 by carlopav