realthunder/FreeCAD_assembly3 #878: .step file parts/sub assemblies need to be kept separated

meprakarsa, 2021-10-15:
Greetings and regards from Batam, Indonesia,

I've been using FreeCAD 0.16, 0.18, 0.19 daily, 0.19 stable, and 0.20 daily (all are main branch).
Just recently been interested to use the "shadow view" /lighttracing provided in this awesome Realthunder branch.

Unfortunately every time I open any .STEP file which contains a lot of parts or sub assemblies, the "body" is only shown one in this branch, can not be expanded. 
Of course i need them to be kept separated since I need to do moving/rotating simulation, animation etc from each part with certain assembly constraints.

Please see following snapshot from one example file:
[image]

OS: macOS 10.15
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +4188 (Git)
Build type: Release
Branch: LinkDaily
Hash: 3de879c7a2ba7111cc769d98636eb6d43189459d
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: C/Default (C)

herrgerd, 2021-11-03:
> 
> 
> thanks a lot, it works! [image]

Hey, if it worked for you, please close your issue soon afterwards, so the bug tracker is emptied a bit :)

meprakarsa, 2021-11-03:
please forgive this newbie who didn't know that issue shall be closed in tracking wise.

closed 2021-11-03 by meprakarsa