realthunder/FreeCAD_assembly3 #880: Toolbars not locking

cadmarx, 2021-10-16:
In Oct 15, 2021 release, Draft Tray and Draft Snap toolbars remain unlocked after a restart.

realthunder, 2021-10-17:
Can you be more specific? Post a screenshot maybe? You know, the Draft tray and snap toolbar can be either in toolbar area or status bar.

cadmarx, 2021-10-17:
> Can you be more specific? Post a screenshot maybe? You know, the Draft tray and snap toolbar can be either in toolbar area or status bar.

Please see the attached screenshot. My experience was with the BIM workbench. Hope this makes it clearer.
[image]
.