realthunder/FreeCAD_assembly3 #882: ok

hamza00789, 2021-10-18:
H f g

realthunder, 2021-10-19:
Which operation did you perform? I did find any command to 'Create shape from mesh'.

hamza00789, 2021-11-19:
> Which operation did you perform? I did find any command to 'Create shape from mesh'.

hamza00789, 2021-11-19:
> Which operation did you perform? I did find any command to 'Create shape from mesh'.

The solution to my problem is in this version of FreeCad. 
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.26155 (Git)
Build type: Release
Branch: master
Hash: 0926a4148bcff11249fd4f56bc30256102ffe105
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US) 
[image]
[image] 

But that doesn't work in this version of FreeCad. 
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United States (en_US) 
[image]
[image] 
 
Can you tell me why?

realthunder, 2021-11-20:
Should be working in the next release which merges with upstream 0.20

luzpaz, 2022-01-12:
@hamza00789 please retest

hamza00789, 2022-01-12:
> @hamza00789 please retest

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git)
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.6.0
Locale: English/United States (en_US)             

There are some issues with the new release. Now even my file is not being imported. Freecad shuts down when I try to import a file.          

https://user-images.githubusercontent.com/92737149/149132004-0de1f5ba-eb1e-4b5f-b099-6fb67537a449.mp4

luzpaz, 2022-01-12:
Thanks for the follow-up. BTW, please use 'Crash' instead of 'shuts down'.

realthunder, 2022-01-17:
@hamza00789, the crash on 01.01 release is a duplicate of #1024. I'll close this first.

closed 2022-01-17 by realthunder