realthunder/FreeCAD_assembly3 #889: [feature request]

anirudhnehra, 2021-10-29:
Hi, great job on this project. Maybe the following enhancements could make it even better.

-Allow to stop/interrupt recompute in case freecad is stuck.
-One improvement in path array tool could be pattern by chordal length(straight distance between consecutive elements on the curve rather than distance along the curve) .This tool will be very helpful to build chains of cycles or excavator /cable tracks. Please see this video for more details. It simulates realistic chain https://youtu.be/KrEIHjZ3Qgs

-Another improvement could be pattern by reference. That means suppose I build array of coordinate systems on a path. Then I make a body and want body to be coincident in position and orientation with each corresponding member of array of CSYS. I could then use attachment constraint of ASM3 and locate first body with the seed member of CSYS array.
Then I just say array on the body and all bodies are created (equal to number of CSYS) and oriented automatically to match the coordinate system array .
This feature of reference array is available in every major cad software. It also helps to reduce memory footprint.
Please see this series https://youtu.be/-R9myIG5XjA
I think Creo Parametric has the best pattern tools worth taking a look at.

-Macro embedding in model tree as a feature. It could write to a CSV file or in report view or create a model tree feature. The main issue is fixing the timestamp at which this macro must be executed using CTRL+R. It must allow to be executed after a feature of choice in the model tree.
-Also, one more improvement could be feature suppression by variable using a spreadsheet.(0 could be feature off and 1 feature on).
-One other improvement could be in sketcher- "Perimeter Dimension". Suppose I have a connected chain of lines/arcs/bsplines in sketch whose Overall length I want to be constant.(Thus overall length is same as perimeter). Then I dimension length of all elements in the sketch except one whose length I want ot be driven by (perimeter dimension-length of all other chain). Maybe It could be done in expressions but we don't have a tool to retrieve length of selected elements in a sketch.
-ARC/bspline length dimension in sketcher can be a good enhancement too.
Though freecad is great, we could always learn something from other software.
Freecad has to become primarily a mechanical design tool and I hope it reaches the same capabilities as the big three(Creo, NX, Catia)

luzpaz, 2021-10-30:
one FR per ticket or else the ticket gets cluttered real fast.

luzpaz, 2022-04-11:
Closing as OP hasn't followed up

closed 2022-04-11 by luzpaz