realthunder/FreeCAD_assembly3 #894: feature request

hamza00789, 2021-11-02:
Blend Corner feature

https://user-images.githubusercontent.com/92737149/139844465-71672ee5-1600-4664-b10e-987bdc70a591.mp4

luzpaz, 2021-11-02:
@hamza00789 Please use less generic ticket summaries so that devs/triager have an easier time understanding what said ticket is. Also it is customary to add some descriptive sentences (besides a screenshot/screencast) of what you are asking for.

hamza00789, 2021-11-02:
> 
> 
> @hamza00789 Please use less generic ticket summaries so that devs/triager have an easier time understanding what said ticket is. Also it is customary to add some descriptive sentences (besides a screenshot/screencast) of what you are asking for.

All the selected faces must belong to the same global support
[image]
[image]
Blend Corner is a tool used to increase the corner radius of any part

herrgerd, 2021-11-03:
This looks very nice, indeed pro software has a lot to offer for its >7000€ price point :) Really looking forward to @realthunder answer here. I _guess_ the fillet and chamfer functions need a lot more basic refinement before adding such advanced features...

czinehuba, 2021-11-04:
A good first start would be just a plain drop-down box for corner type.

[image]

realthunder, 2021-11-09:
I suppose this function can be somewhat achieved by exposing OCCT's multi-radius fillet capability. Part's fillet can specify two radius per fillet edge, but according to OCCT [API](https://dev.opencascade.org/doc/refman/html/class_b_rep_fillet_a_p_i___make_fillet.html#a5b139b756f74e55df5599bd6acea2a45), it actually supports more than two per edge. I'll give it a try some day.

[image]

hamza00789, 2021-11-09:
> I suppose this function can be somewhat achieved by exposing OCCT's multi-radius fillet capability. Part's fillet can specify two radius per fillet edge, but according to OCCT [...]

Multi-radius fillet quite contrasting frome the Blend Corner  
I Know Blend Corner is a tool used to increase the corner radius  
But the bland corner is used in many other places  
It also works for more than three sections
For Example

https://user-images.githubusercontent.com/92737149/140958240-50f9a6e3-c8de-4c01-98b0-d1972d4c1db1.mp4

hamza00789, 2022-01-18:
> I suppose this function can be somewhat achieved by exposing OCCT's multi-radius fillet capability. Part's fillet can specify two radius per fillet edge, but according to OCCT [...]

If you want to hear, I have an opinion. As I said  Blend Corner works on more than 3 or more corners. I have also requested a variable radius. This is different from [multi-radius or variable radius]. Of course, you are a developer, you know more and more about it than I do. if you are looking for a function like Blend Corner. You may need three commands that are already in the freecad. As you can see in the picture icons.Their names blend curve, isoCurve, and Filling. This feature can be achieved by using these three things if you look closely.   

The filling command can be used on any curvature area if you know. And the ISOcurve of any curvature area also comes out (horizontal, vertical). If you think this way you will find that the Blend corner can be used on more than three corners and also on different curvature areas.

hamza00789, 2022-01-20:
> I suppose this function can be somewhat achieved by exposing OCCT's multi-radius fillet capability. Part's fillet can specify two radius per fillet edge, but according to OCCT [...]

Would you like to see this?[blend.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7908630/blend.zip)   
Example built inside the upper post commands.

hamza00789, 2022-01-20:
> I suppose this function can be somewhat achieved by exposing OCCT's multi-radius fillet capability. Part's fillet can specify two radius per fillet edge, but according to OCCT [...]

Will you watch it again from here?      
https://youtu.be/K6UoGduJo6g?t=152

hamza00789, 2022-01-29:
> 
Multi-radius fillet and Blend Corner 

https://www.youtube.com/watch?v=AhMmb2Mm4p8

hamza00789, 2022-01-29:
https://grabcad.com/tutorials/edges-fillet-with-blend-corner

hamza00789, 2022-02-08:
https://forum.freecadweb.org/viewtopic.php?f=8&t=66038

hamza00789, 2022-02-09:
[mp4.zip](https://github.com/realthunder/FreeCAD_assembly3/files/8036004/mp4.zip)

realthunder, 2022-02-09:
I'd like to work on this corner blending feature, as well as the multi radius fillet. The thing is, after seeing how other CADs are doing it, I think these two operation definitely require at least some on screen push and pull support, which currently FreeCAD lacks. So I'll need to work on those first, probably not in the immedate future, though. Too much other stuffs are pending.

hamza00789, 2022-02-09:
Thanks for working on them. My best wishes are with you.

luzpaz, 2022-08-13:
@hamza00789 please remove proprietary references in the FR