realthunder/FreeCAD_assembly3 #896: [Improvement] Set size of a new datum plane relative to 3D view and zoom level, if created in an empty body

Rabbit-sk, 2021-11-02:
Hi RealThunder,

size of a new datum plane created in a body, which already contains some features, is adjusted with respect to their size. But if a scene contains objects with large dimensions (meters), the size of a new datum plane is too small, if it is created in a new empty body.

https://user-images.githubusercontent.com/74265320/139944007-eab17ef7-2eec-4b43-abfc-8ad2ad818ca3.mp4

Getting a visual feedback of adjusted position of such datum plane involves a lot of zooming, and often it is impossible to find small datum plane in the large scene after changing its offset. 

Autosizing of a new datum plane could be improved. Setting new datum plane size with respect to the bounding box of the whole scene wouldn't be good, because if scene is zoomed and not all objects would be in the 3D view, the new datum plane could be bigger than 3D view. 

If a new datum plane will be created in an empty body, its size could be calculated from zoom level and / or 3D view height, as 3D view width is probably allways bigger than height. This relative size could be 50% of 3D view height, or its setting could be placet in preferences.

Best regards

OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git)
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: Slovak/Slovakia (sk_SK)

Rabbit-sk, 2022-01-10:
Hi @realthunder !
Is there anything new about this?

SteveSwaringen, 2023-04-02:
Piling on. Would love to see this feature. Or even the ability to manually adjust the visible size of the plane after it has been created. The parameters appear to be locked.