realthunder/FreeCAD_assembly3 #897: [Issue] Some strange issue with Asm4 and new renderer

ngocminhdao88, 2021-11-02:
I have some weird issue when using Asm4 workbench with the new renderer. I had this issue already from 2 or 3 previous releases ago.

Right result when using old (auto) renderer.
[image]

Wrong result when using new renderer. Also the parts are jumping around in assembly.
[image]

Here are my working file.
[adxl345_carrier.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7463931/adxl345_carrier.zip)

And here is my FreeCAD information:

> OS: Ubuntu 20.04.3 LTS (i3/i3)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

luzpaz, 2021-11-03:
CC @Zolko-123 (just giving you an FYI)

Zolko-123, 2021-11-03:
> I have some weird issue when using Asm4 workbench with the new renderer.

you are using LCS that are not attached (MapMode not set correctly) in their respective parts (Bodies). Also, you don't "Refine" your parts (Bodies) which I think is bad idea. All this might lead to unexpected results.

I suggest you try your assembly by setting your parts to "Refine"=True, and  attaching correctly all your LCS (MapMode and Support should be set). I then please report.

ngocminhdao88, 2021-11-03:
Hi Zolko,

i still have the same issue after refining the part/body and mapping the LCS to origin. Switching between two renderers some how breaks the assembly.

New sample here.
[adxl345_carrier.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7469316/adxl345_carrier.zip)
.

closed 2021-11-20 by realthunder