realthunder/FreeCAD_assembly3 #899: Sketching Planes

satiutni, 2021-11-04:
I have created a file with a new part and new body in Master branch.
I then go to create a new sketch on the XZ plane. When editing this sketch I am presented with the XZ plane for sketching and the cube indicates I am sketching on the front. 

I then create another new file with a new part and new body in ASM3
I go to create the same sketch on the XZ plane. When I attempt to edit this sketch I am presented with the XY plane for sketching and the cube indicates I am sketching on the top.
If I rotate the cube to sketch on the front (XZ plane) I am unable to create any elements in  the sketch.

Is this expected ASM3 behaviour? Of course I could do all my sketching with the XY plane and then apply features to other planes but when following a tutorial that calls for using the XZ plane for the initial sketch I cannot do it.
When I look at the attachment for the (empty) sketch it is definitely attached to the XZ plane.
How do I resolve this? And how is the difference in behaviour between Master and  ASM3 explained.
I know when I come to trying this tutorial with my students the questions I am asking now will arise but I have no answer for them at present.
I have attached the following files:
Mechanical Part1.fcstd which was created using ASM3 and has only been opened in ASM3.
Mechanical Link.fcstd created in Master and only opened in Master.
I am using :
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/Australia (en_AU)
and
OS: Linux Mint 19.3 (X-Cinnamon/cinnamon)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24291 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.19.2)
Hash: 7b5e18a0759de778b74d3a5c17eba9cb815035ac
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/Australia (en_AU)
respectively on
Linux Mint 19.3 Cinnamon 4.4.8 Kernel 5.4.0-81-generi
[Mechnical Examples.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7471728/Mechnical.Examples.zip)

satiutni, 2021-11-04:
I have since tried this with Windows 10
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.717.24301 +3627 (Git)
Build type: Release
Branch: LinkStage3
Hash: f63a5ea7b459872a3798027df0ff25e776524859
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/Australia (en_AU)

The behaviour of ASM3 is still the same.
After choosing any plane, other than XY plane, to start a sketch only the XY plane is presented for the sketch and it is not possible to select another plane to sketch on even when the map mode for the sketch is one of the other standard planes.
Eg. I choose XZ as the plane of attachment, the data for the sketch shows that the map mode is XZ but the only plane that a sketch can be made on is XY.

luzpaz, 2022-01-22:
@satiutni can you still recreate on latest Daily?

mwlandolsi, 2022-02-24:
Same problem.
Just create new body on new document, the click on sketch icon and choose XZ plane. draw then a sketch on the given plane. The sketch is created on XY and not on XZ as selected in the beginning.

OS: Linux Mint 20.1 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 2022.111.30764 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: ef3092bcbca5475eff4eb7ff7933eb3e9bbbe4b9
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

mwlandolsi, 2022-02-24:
Looks like a problem of configuration files. After deleting *.cfg files and "webdatapersistant" directory in /home/userName/.FreeCAD directory, then I create a sketch on XZ plan, and it is OK.