realthunder/FreeCAD_assembly3 #900: Part Workbench - Create Shape from mesh freezes Freecad

tproulund, 2021-11-04:
Hallo

I am using the Appimage version 2021.922 for Arch Linux and noticed that the software freezes when I try to use the part workbench to create a shape from a STL and eventually I just have to force quit the software. 

I tried going back to the official 0.19 and there it works fine, though a bit slow which is properly due to the size of the STL

luzpaz, 2021-11-04:
please paste the full [About](https://wiki.freecad.org/About) info in to the first post.  Also can you attach the STL?

tproulund, 2021-11-04:
Sorry about that, here is everything 

The STL is this one https://www.thingiverse.com/thing:2854173 

 OS: Arch Linux (GNOME/gnome)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.922.24301 +3888 (Git) AppImage
Build type: Release
Branch: LinkStage3
Hash: 399863d7e37b354fea6687767605e661beace504
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

marthinwurer, 2021-11-16:
I've used FreeCAD to slice some large STLs and while it does freeze, it's still working (terribly inefficiently) in the background while it looks like it's frozen. It's a performance issue that leads to this functionality issue, but I don't know where that stems from.