realthunder/FreeCAD_assembly3 #904: missing keyboard shortcuts.... && memory hogging/leak

apper002, 2021-11-06:
missing keyboard shortcuts...

and

https://forum.freecadweb.org/viewtopic.php?f=3&t=63499

OS: Ubuntu 21.04 (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/United States (en_US)

apper002, 2021-11-06:
removed .FreeCAD, restart gave keyboard shortcuts.

apper002, 2021-11-06:
memory problem still there

luzpaz, 2021-11-07:
Please update ticket name and content if that's the case

apper002, 2021-11-08:
updated

adding some camfer res mem got upto 6G...................

realthunder, 2021-11-08:
Is it possible to share your model somewhere? You can send me email at realthunder.dev at gmail.

realthunder, 2021-11-10:
I have tried your file, and I don't think there is any major memory leak (can't say much about small ones). Most of the memory is held up in the undo stack. You can keep your memory usage in check by limiting the undo stack size as shown below.

[image]

In case you are curious, here is the process of proving my claim of no major memory leak. I am not sure if you have enabled `Enable undo/redo on recompute` option (also shown above). This option will allow FreeCAD to undo explicit recompute (i.e. when you hit that `Refresh` button). Please enable this option which is required in the following steps.

After a refresh start, load your file, I am seeing 2.5GB memory used. Right click the document icon in the tree view, select `Mark to recompute`, and then hit recompute. This will perform a full document recompute. I got an error in one of the sketch complaining missing external geometry. Just re-attach the edge as shown below. Then hit recompute again. This time it recompute all the way till the very last fillet feature, complaining about command not done. I didn't bother to fix it. Now the memory usage on my computer is 4.2GB. This indicates that a whole document recompute requires about 1.7GB, which comes from dumping the old model data to the undo stack.

https://user-images.githubusercontent.com/1207888/141083927-4826682d-99e1-4d6f-8a47-63db1d8e4f60.mp4

Now, close the document WITHOUT saving it. You'll find that the memory usage does not shrink. This can be explained by operating system memory management policy. For allocated memory pages, the OS will only actively reclaim the memory pages if the whole system is under memory stress. You can prove this by re-open you document, then repeat the above process, and observe that the memory usage stays the same. You can do one more full document recompute, and see the memory usage grows by about 1.xGB.

Another way to prove my point, is to disable the `Enable undo/redo on recompute` option or just disable `Using undo/redo on documents` (which disables undo entirely). And you'll find the memory usage stays about the same on repeated full document recompute.

apper002, 2021-11-11:
Okay, i clear both the undo/redo buttons ( i had it already at two, but now it is completely disabled).

Model recomputes fine, have no errors.
Initial mem usage still 2.6G (but seems we need to live with such mem usage).
[image]

I add **1** fillet, mem usage jumps with 300MB
[image]

So now added a couple of fillets and mem size still the same, that's good
[image]

So no redo/undo but mem size sofar too big and time consuming but stable.

Thanks again for your time Zheng!!

apper002, 2021-11-20:
Well back at it again....

Upto sketch number 57 with different model. f
From sketch 50 to 57  res mem size jumped from 1G to 1.6G and dramatically slowing down.

And no undo/redo whatsoever active...

Is this Freecad or the underlying Cascade engine ?

Rgds
Apper

realthunder, 2021-11-21:
Maybe share your file?

Anyway, there is another trick you can try. You can split  your model history into multiple files using PartDesign boolean feature. The following screencast shows how. Create a new file and save it first. Then creates an empty body. Select the feature in the old body where you want to continue modeling. Then, switch back to the new file, and add a boolean feature. The feature from the old body is brought in using a `SubShapeBinder` which is capable of partial loading.

The screencast first shows what happens without enabling partial loading. When you close all files and open the file with the new body, the old body file will be opened automatically. Once you enable the partial loading by setting the `PartialLoad` property, you will be able to continue modeling without the old body. And toggling the `PartialLoad` property can bring back the old body.

Note that there is no limit on where you can split the history. It is perfectly fine to split in the middle, which is effectively a fork. Or, you can boolean the entire old body, so that any new feature you add to the old body will be auto reflected to the new body.

https://user-images.githubusercontent.com/1207888/142755646-8b66e72b-9013-46f8-8248-3359f2d75db6.mp4

apper002, 2021-11-22:
Hi Zheng,

 

Once again thanks for all your time and effort!!

 

Tried it, works at first.

 

Unfortunately: i start from say 600MB and after adding three Sketches i am at

1.4GB again and slowing down rapidly.

 

20276 apper     20   0 2593440 599016 235772 S   2,0   2,9   0:40.77 FreeCADLink

0276 apper     20   0 3292868   1,4g 243156 S 105,5   7,1   3:05.87 FreeCADLink

 

https://drive.google.com/file/d/1TDwllCeEZd7WyRJe-gZcGrNkpcJhwR3V/view?usp=sharing

 

 

 
 

Gesendet: Sonntag, 21. November 2021 um 10:44 Uhr

 

https://user-images.githubusercontent.com/1207888/142755646-8b66e72b-9013-46f8-8248-3359f2d75db6.mp4

apper002, 2021-11-22:
PS.

Also tried all "Bind Modes": Synchronized/Frozen/Detached

 
 

Gesendet: Sonntag, 21. November 2021 um 10:44 Uhr

 

https://user-images.githubusercontent.com/1207888/142755646-8b66e72b-9013-46f8-8248-3359f2d75db6.mp4

apper002, 2021-11-22:
PPS.

 

Saving as step file, create new model and reimport and sofar the Mem usage stays at 651MB

 
 

Gesendet: Sonntag, 21. November 2021 um 10:44 Uhr

 

https://user-images.githubusercontent.com/1207888/142755646-8b66e72b-9013-46f8-8248-3359f2d75db6.mp4

apper002, 2021-11-23:
Any idea why this exporting and importing makes a difference?

realthunder, 2021-11-24:
Can't seem to download your last shared file. It ask me to request access first. I've requested.

apper002, 2021-11-24:
https://drive.google.com/file/d/11FwFdS11vitCK4FFhJQjwMx5v1nKnclS/view?usp=sharing

 
 

Gesendet: Mittwoch, 24. November 2021 um 07:07 Uhr

realthunder, 2021-11-24:
This is what I get

[image]

apper002, 2021-11-24:
https://drive.google.com/file/d/11FwFdS11vitCK4FFhJQjwMx5v1nKnclS/view?usp=sharing

 

 

 

Gesendet: Mittwoch, 24. November 2021 um 14:13 Uhr

apper002, 2021-12-31:
still an issue
not shure what todo or what's with the problem with 50 or more sketches...

luzpaz, 2022-03-18:
@realthunder the file is now available, jfyi