realthunder/FreeCAD_assembly3 #906: Solid feature using but show only surface

hamza00789, 2021-11-07:
[image]
[image]
FreeCAD-asm3-Daily-Win64-Py3-Qt5-20211015

herrgerd, 2021-11-07:
Can you upload the model? Looks a bit strange...

hamza00789, 2021-11-08:
> 
> 
> Can you upload the model? Looks a bit strange...

[123.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7495207/123.zip)

herrgerd, 2021-11-08:
I can confirm this in the most recent daily build.
In .1015 and .0716 it works as expected!

hamza00789, 2021-11-08:
> 
> 
> I can confirm this in the most recent daily build. In .1015 and .0716 it works as expected!

I don't understand In .1015 and .0716 it works as expected! please Explain

realthunder, 2021-11-09:
Please change `Preferences -> Display -> Render caching` to `Experiemental`, before activating sketch clip plane.

Note that the two body you are trying to clip is separate solid. So you will only able to see solid sections and no hollow inner body. I have no idea how Catia can guess you are trying to do a subtraction and display the correct result. Or maybe you didn't show the full tree view, and it has already been cut?

If you really want to do this before cut, you can set a different color for one of the body, although you'll only be able to see the colored section in certain angle.

https://user-images.githubusercontent.com/1207888/140839015-17f04e00-b21e-4bff-bc59-c927c41785d3.mp4

hamza00789, 2021-11-09:
> 
> 
> Please change `Preferences -> Display -> Render caching` to `Experiemental`, before activating sketch clip plane.
> 
> Note that the two bodies you are trying to clip are separate solids. So you will only be able to see two overlapping solid sections and no hollow inner body. I have no idea how Catia can guess you are trying to do a subtraction and display the correct result. Or maybe you didn't show the full tree view, and it has already been cut?
> 
> If you really want to do this before cut, you can set a different color for one of the body, although you'll only be able to see the colored section in certain angle.
> sketch-section.mp4

I have show the full tree view         
When there are two solid overlaps in the catia, the difference between the two can be seen from the clipping plane with out using cut command 
 
Two solid using cut command in FreeCad  
[image]
 
Two solid Without using cut command in Catia
[image]

herrgerd, 2021-11-09:
I don't exactly know what you're creating there. But that's not how engineering works and I'd call it an "the software has to deal with your style of doing things" situation.

Of course the bodies have to be solid when in clipped plane view, but the program can't assume the inner part to be "empty" just because it is in some other body!

hamza00789, 2021-12-01:
> I don't exactly know what you're creating there. But that's not how engineering works and I'd call it an "the software has to deal with your style of doing things" situation.
> 
> Of course the bodies have to be solid when in clipped plane view, but the program can't assume the inner part to be "empty" just because it is in some other body!
> 
> The issue should be closed because the one issue is solved by changing the renderer and the other thing isn't an issue but "bad" usage of the program. 

Don't be angry, listen to me once and see. The way it works in the catia makes it very easy to work in the sanitary field.  It is much easier if you know about this field of sanitary. It shows where the water is coming from and where it is going.  This will make it much easier for those working in the sector. I know a lot about the sector. My main connection is with the sector. I think in free cad it should have been the same as in catia. Catia is software used in many field. Catia the solid shows it in such a way that if it comes on top of each other, it shows empty from the inside. Of course this is a very useful thing that happens in the catia. Free cad doesn't know about it

realthunder, 2021-12-02:
It is certainly possible to do, because other software have done it. I guess what they have done is to use a dedicated pass for rendering section fills. Right now what I've done is to render the fill for each solid separately.

The basic principle of rendering section fill is to use a [stencil buffer](https://learnopengl.com/Advanced-OpenGL/Stencil-testing). To make this effect work, we need to render all solids with stencil activated first, then check the stencil buffer content and fill the area that are occupied by only one solid, and skip any overlapped area.

I would give it a try, sure. But no promise on when. Too many things pending right now.

hamza00789, 2021-12-05:
> It is certainly possible to do because other software has done it. I guess what they have done is to use a dedicated pass for rendering section fills. Right now what I've done is to render the fill for each solid separately.
> 
> The basic principle of rendering section fill is to use a [stencil buffer](https://learnopengl.com/Advanced-OpenGL/Stencil-testing). To make this effect work, we need to render all solids with stencil activated first, then check the stencil buffer content and fill the area that is occupied by only one solid, and skip any overlapped area.
> 
> I would give it a try, sure. But no promise on when. Too many things pending right now.

[image] 

[image] 

file         

[123.zip](https://github.com/realthunder/FreeCAD_assembly3/files/7654978/123.zip)

closed 2021-12-29 by realthunder

hamza00789, 2022-01-12:
thanks to solve