realthunder/FreeCAD_assembly3 #908: [FR] Positioned sketch

hamza00789, 2021-11-08:
Creating a positioned sketch enables you to define (and later change) explicitly the position of the sketch absolute axis. 

https://user-images.githubusercontent.com/92737149/140731157-4d01980f-56a8-4a28-906e-a9a4d9013665.mp4 

profile (2D geometry of the sketch) remains unchanged (even if the sketched profile is under-constrained)

https://user-images.githubusercontent.com/92737149/140731246-72a87047-5203-45df-b195-04211ffd7eab.mp4

realthunder, 2021-11-09:
Aren't these transformation already supported by Sketch attachment? Apart from flip/mirror, I suppose that's what you want? You can flip the sketch using a Link with -1 scale, and you can edit it as flipped through the Link. Link is fully supported in Part workbench, but not PartDesign body, yet. Maybe I'll support it through sketch export.

https://user-images.githubusercontent.com/1207888/140898734-71332bfb-dfdc-4576-813e-92c304b3bfa6.mp4

hamza00789, 2021-11-09:
> 
> 
> Aren't these transformation already supported by Sketch attachment? Apart from flip/mirror, I suppose that's what you want? You can flip the sketch using a Link with -1 scale, and you can edit it as flipped through the Link. Link is fully supported in Part workbench, but not PartDesign body, yet. Maybe I'll support it through sketch export.
> sketch-flip.mp4

Thanks for reporting the issue. But it's a little bit different.

realthunder, 2021-11-10:
That's why I mention the sketch `Attachment` functionality. If you are not satisfied with the default attachment, you can edit it. With my branch, just right click the sketch and select `Attachment editor` in the context menu. It is also available in application menu `Part -> Attachment`. There is also a `Flip` option there, but it won't mirror the sketch. It is just to allow you to attach to the other side of the face. Using my branch, you can simply rotate the sketch while editing to edit it from the other side. You need to hold `Alt` to enable sketch rotation.

[image]

https://user-images.githubusercontent.com/1207888/141030500-3866e4ce-cd9d-479b-aad7-e167f8737cb2.mp4

closed 2021-11-10 by hamza00789