realthunder/FreeCAD_assembly3 #909: link problems when using subshape binder and assemblies to link to another file...??

bambukouk, 2021-11-09:
[image]
I have two part files A and B. Each contains bodies. In the case of B these bodies are created using sub-shape binder to the body in part A. Both the bodies in A and B are linked to another assembly file.
No problem at all, it works fine, but...

If I try to be clever and group bodies in part B in an sub-assembly, and then link both the sub-assembly in B and body in A to the top level assembly it all goes to pot, complaining about not being able to get the sub-shape binder (I think).

Is it me being thick again? or have I stumbled upon a problem?

realthunder, 2021-11-09:
Can you please post the document with the error?

bambukouk, 2021-11-09:
Thank you.
I will create simplified example.

bambukouk, 2021-11-09:
Can't recreate using simplified example (so probably my mistake, as usual),
 but I have kept original "problem" folder, so if you can make any sense of it?

The error message I get, is
[image]

[sub_problems.tar.gz](https://github.com/realthunder/FreeCAD_assembly3/files/7505864/sub_problems.tar.gz)

realthunder, 2021-11-10:
There is indeed a bug in `SubShapeBinder` restoring logic, and is fixed by the above referenced commit.

As for using `Part` vs `Assembly` container, asm3 supports using almost any type of objects are an assembly component. The advantage to wrap the component as a sub-assembly is to allow you to define component interface through [Element](https://github.com/realthunder/FreeCAD_assembly3/wiki/Concepts#element). For example, check out [this](https://github.com/realthunder/FreeCAD_assembly3/wiki/Replacing-Part) example on how to make interchangeable components.

bambukouk, 2021-11-10:
@realthunder `There is indeed a bug in SubShapeBinder restoring logic, and is fixed by the above referenced commit.`

Thank you! it's a relief that you have actually found something (rather than it being yet another case of my confusion...)

I understand advantages of using `Assembly` container (thank you for the links), but there are also disadvantages - namely added complexity. My assemblies are large, containing numerous objects (is there a way of querying FreeCAD file to check number of objects?) which is why I am moving away from single file design to multi/linked file design.

I am using Assembly 3 elements and constraints only for `"kinematic"` objects.
For `"static"` objects I prefer to avoid using elements and constraints, instead using placement (in relation to other object or main datum).

It was all very fine and OK when design was simple and with reasonable number of objects, but for large complex designs I am getting swamped :-)

I would love to have a discussion with you to try and better understand these things :-)
but this is not the place for it.... plus, of course your time is better spent in what you are doing for us all - thank you again!

realthunder, 2021-11-10:
No problem. On the contrary, assembly container offers some functionality to handle the complexity. For example, you can use the assembly `Freeze` function to strip out shape history inside the body. Checkout the tutorial [here](https://github.com/realthunder/FreeCAD_assembly3/wiki/How-to-Handle-Large-Scale-Assembly). But of course, it also brings some inconvenience when you actually want to change the sub-assembly.

bambukouk, 2021-11-10:
Yes `jpg87` has helped me a great deal (both personally and through his webpage) when I was starting on my Assembly 3 journey.
Having read all this, I remain puzzled as to what advantages are there to using assembly container for objects which will never be used in any assembly constraints?
These are the objects that I refer to as `"static"` - once they are placed in correct position (relative to datum and other `"real assembly/kinematic" `objects) there is no need to constrain them?

realthunder, 2021-11-10:
The `Freeze` function will create a snapshot of the current state of the assembly and leaving all invisible objects, that include all the intermediate model shapes, sketches, datum, or constraint/element in lower sub-assembly. Isn't that what's bothering you? I mean you mentioned about your concern of object count?

When the frozen assembly (stored in a separated file) is added to another assembly in a different file, the next time you open that assembly file, it will only partially load the frozen assembly file. Those invisible objects will not be loaded, saving both loading time and memory space. Now think about the compounding effect where those sub-assembly have their own sub-assemblies and so on.

bambukouk, 2021-11-10:
That might be precisely what I am looking for....
Will have to experiment with freezing of sub-assemblies in their own files (used in top level linked assembly).

So, let's say I have sub-assembly in it's own file. It is frozen.
It only needs to be statically constrained i.e. anchored somewhere when used in top level linked assembly file.

I presume `"locked" constrain` would take care of it? if so, how do I use "frozen" element to constrain it?

or... do I `"lock constrain"` it in the original sub-assembly file? 

Thank you - time to start experimenting and learning!

realthunder, 2021-11-10:
You need to have at least one `Element` in the sub-assembly before frozen. You can always unfreeze it and create a new element. Once added to the upper assembly, select that `Element`, and click the `Lock constraint` icon to lock the sub-assembly.

bambukouk, 2021-11-10:
Perfect!
so if I have dozen objects (eleven of which are linked copies) in the sub-assembly, I only need to create one element on the original object, to use it to lock sub-assembly in the assembly linked file?

I think we can almost close this issue with this comment! ?
Thank you - you have been great help to me today 
I feel I am making progress!!

bambukouk, 2021-11-10:
Have started and just a couple of parts...
Treeview looks like pic below.
Imagine what it will be like by the time I have hundreds of sub-assemblies
Mind boggles :-) :+1: 

Is there a way to make treeview more manageable?

[image]
s

bambukouk, 2021-11-10:
Seem to have discovered another problem? (should I start separate issue?)
As you can see from pic below there are some problems carrying over / linking element labels?
There used to be similar problem in the past #599
Don't know if this is related?

[image]

bambukouk, 2021-11-10:
@realthunder 
`The Freeze function will create a snapshot of the current state of the assembly and leaving all invisible objects, that include all the intermediate model shapes, sketches, datum, or constraint/element in lower sub-assembly. Isn't that what's bothering you? I mean you mentioned about your concern of object count?`

Yes, it is indeed working very well and as I wanted (re object count and treeview complexity):

[image]

There is something odd about axlebox sub-assembly that not only causes problems with element label not updating up the hierarchy, but also some strange behavior with parts list??? will have to see what it is?

closed 2021-11-20 by realthunder