realthunder/FreeCAD_assembly3 #910: freecad freeze during zoom out

anirudhnehra, 2021-11-09:
Hi,
I ma using latest freecad Linkstage3 version. Whenever I load a big step assembly and zoom out too far, the freecad window freezes and the processor is busy all the time. Please see the attached screen capture.

https://user-images.githubusercontent.com/42260300/140973815-16978c48-bc08-4363-a02d-444591bb8590.mp4

here is my freecad version info:
OS: Linux Mint 20.2 (XFCE/xfce)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 2021.1015.24301 +4280 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 556c87868ea46796242156e9f73eae98259794b7
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.2
Locale: English/India (en_IN)

Thanks and Regards 
Anirudh

realthunder, 2021-11-09:
Depending on the complexity of your STEP file, when zoomed out too far, the pick radius (with size in screen coordinate) may pick too many geometry elements (because they are so far out and therefore very small on screen) that your system can handle.

The problem is normally less obvious, because the software will only keep the nearest picked element while traversing the scene. The freezing problem is probably caused by the fact that you have enabled `Picked list` here, which will store all picked geometry elements, each with a complete scene graph path, causing explosion in memory usage. So, please turn off `Picked list` while working on complex geometries. You can use `Pick geometries` command instead (shortcut `G, G`) to pick hidden geometries.

anirudhnehra, 2021-11-10:
Yes that solves the issue.

Thanks and Regards
Anirudh Nehra

closed 2021-11-10 by anirudhnehra

anirudhnehra, 2022-03-03:
Hello,
Your branch literally sucks when I do serious part design. I just cannot
embed other workbench features especially curves workbench in your part
design using wrap feature. If I try wrapping a curve into part design I
just can't use it for additive pipe as a path. Also when your experimental
renderer is used it randomly shows coordinate systems created in part
design somewhere else.
There are tons of warnings about out of scope stuff on every command. I
still don't understand what link out of scope means. Pls fix these simple
things first before trying something else.

anirudhnehra, 2022-03-03:
> Are these requests or demands? If they demands, then you'll need an
> attitude adjustment. If they are requests, you'll need an different
> approach to requesting your needs in the future (AKA 'attitude adjustment').

Sorry for this way. I take back my words. Sometimes frustrating it gets.

luzpaz, 2022-03-03:
If RT is cool with you talking that way to him, better to keep it in a DM since it looks really bad in an issue queue and other users could get an incorrect impression of RT (or of you (if you don't want to misrepresent yourself))

anirudhnehra, 2022-03-04:
Hey I am sorry again. Now let's forget about it.
If my wordings are sent to every user of freecad I am sorry again to rt and
to them.

realthunder, 2022-03-04:
I am okay with the complaint, and I can understand the frustration. But what you are posting is more like ranting without any elaboration on the actual issue (actually two issues, posted in an irrelevant thread). That's not going to help solving any problem. Please make a proper issue report, or two.